mavlink/mavros #942: Problem Using mocap_pose_estimate with APM (Indoor Localization with UWB)

mellonUAV, 2018-02-07:
Hello,

I am trying to use Erle Brain 3 to fly UAV indoors using UWB (Ultra wideband). Ultra wideband will give the position of the UAV (x, y, z) and I will need to inject this into Ardupilot. I am trying to achieve this by using Mocap Pose Estimate. I white listed this plugin which says it is initialized in the roslaunch terminal. I am publishing to /mavros/mocap/pose topic using the following ros node:
```

#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/OverrideRCIn.h>

int main(int argc, char **argv) {

  ros::init(argc, argv, "position_sender");
  ros::NodeHandle n;
  ros::Publisher pub = n.advertise<geometry_msgs::PoseStamped>("mavros/mocap/pose", 10);    
  ros::Rate rate(20.0);

  while(ros::ok()){

		geometry_msgs::PoseStamped msg;

		msg.pose.position.x = double(5);
		msg.pose.position.y = double(2);
		msg.pose.position.z = double(-3);

		pub.publish(msg);
                ros::spinOnce();
		rate.sleep();
	}	
}
```

I connect Erle and my PC using ROS MASTER URI. I can read the topic from both ERLE and my pc using "rostopic echo /mavros/mocap/pose". I am trying to see the result in APM Planner v2.0.24 in ATT_POS_MOCAP message but nothing is appearing here. 

To check if I am properly connected the Erle Brain via ROS I use:

> $ rosservice call /mavros/set_stream_rate 0 10 1
> $ rostopic echo /mavros/imu/data_raw

which seems to be correctly receiving the IMU data. 

I try to do this on my Erle Brain as well but when I try to set mavros stream rate I get this:

> rosservice call /mavros/set_stream_rate 0 10 1
> No handlers could be found for logger "rosout"
> ERROR: Unable to communicate with service [/mavros/set_stream_rate], address [rosrpc://melih-N552VW:35985]

Do you guys have any idea where I am doing wrong or missing something. Any help is much appreciated!

### MAVROS version and platform

Mavros: 0.18.3 (Erle Brain-3) (I am not sure how to properly check this. They say this is the generic version in Erle Brain but I am not sure. I installed mavros to PC by "sudo apt-get ...")
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: V3.5.0-rc9 (242c8f0b)

### Node logs

```
~/erle_ws$ roslaunch mavros apm.launch fcu_url:=udp://10.0.0.2:6001@ gcs_url:=tcp-l://
... logging to /home/melih/.ros/log/1d52e220-0bed-11e8-9670-d017c21fd173/roslaunch-melih-N552VW-11985.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

[... 401 lines cut ...]
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: udp://10.0.0.2:6001@
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
```

### Check ID

```
rosrun mavros checkid
OK. I got messages from 1:1.
```

TSC21, 2018-02-07:
Does APM support `ATT_POS_MOCAP` msg?

mellonUAV, 2018-02-08:
@TSC21 This is my first time working with APM or any other FC so I a am noob :) I am not sure how to check if APM supports it or not. I think it supports because I managed to see ATT_POS_MOCAP once in APM Planner but no matter what I try, I didn't manage to make it work again. I think I just changed something I foun on the interwebs and forget what.

As I sad, I am not sure if APM supports Mocap Pose Estimate plugin, how can I check this?

If it doesn't support, is there any other method I can use to localize the UAV with a position data?

Thank you!

khancyr, 2018-02-08:
Hello,

Currently, ArduPilot doesn't use this message

TSC21, 2018-02-08:
You can use `fake_gps` plugin as an option.

mellonUAV, 2018-02-15:
@khancyr @TSC21 

I went with HIL_GPS message to localize in indoor environment as you guys helped. Thank you so much! This issue can be close I guess.

closed 2018-02-15 by TSC21