mavlink/mavros #943: MAV_CMD topic?

myeabhishek, 2018-02-08:
Hi, I am using MAVROS on RPI with PX4 on pixhawk 2. I want to get a callback each time a MAV_CMD is received by the flight controller on ROS. 

I looked in the documentation, however, I was not able to find a relevant topic for it. Is there a topic available for this or will I have to use "mavlink/from" topic and then parse accordingly?

vooon, 2018-02-08:
Thire is no topic for that. And what exactly you trying to receive from FCU?

myeabhishek, 2018-02-08:
I am trying to get camera commands that are being sent from the QGC to flight controller. My camera is connected with RPI. So based on that command I'll operate the camera.

myeabhishek, 2018-02-08:
There is a uORB topic called "vehicle_command" on PX4. Basically, I want to have that data on my companion computer.

vooon, 2018-02-08:
That is impossible. Mavros only reacts on messages coming from autopilot.

myeabhishek, 2018-02-09:
Can these commands be routed to mavros by anyway?

vooon, 2018-02-09:
You may try to setup intel's mavlink-router, but i'm not sure.

lamping7, 2018-02-09:
I'm not sure of your exact application, so I'm not sure if this is helpful, but as an alternative option, you can connect an AUX from your FCU to the RPi and take a picture when the RPi gets the trigger pulse in it's GPIO.

myeabhishek, 2018-02-10:
@vooon Can you please tell me more about the mavlink_router? Is it possible for me to get all mavlink traffic coming from QGC onto companion computer via using this?

TSC21, 2018-02-10:
Better ask there https://github.com/intel/mavlink-router. This is out of scope for the topic. But expect in the near future to be able to get uORB topic data directly on MAVROS with RTPS. Closing

closed 2018-02-10 by TSC21

lamping7, 2018-02-10:
https://github.com/intel/mavlink-router

It's what the Intel Aero uses