mavlink/mavros #944: Crashes when building source for Ubuntu 16.04 kinetic

chngdickson, 2018-02-09:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Trouble building workspace for mavros while building with source installation. I followed this guide. https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation 
The problem i come across is when i start building my workspace with (catkin build -j1 -l1). As per tutorial, 6 packages were found when building catkin workspace. It was installing fine until i started building mavros. The screen just freezes and my raspberry pi just stops functioning. There is no error code/

In addition, i cant even use ssh to (sudo shutdown -h now)  the system.  It doesn't seem to work.

Can any of u kind soul please help me out.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ -] PX4

Version: ?3.7.1?

### Node logs

Here's the output when it crashes
Profile: Default
Extending: /opt/ros/kinetic
Workspace: /home/pi3/catkin_ws

Source Space: [exists] /home/pi3/catkin_ws/src
log Space: [missing] /home/pi3/catkin_ws/logs
Build Space: [exists] /home/pi3/catkin_ws/build
Devel Space: [exists] /home/pi3/catkin_ws/devel
install Space: [unused] /home/pi3/catkin_ws/install
DESTDIR: [unused] None

[build] Found '6'packages in 0.0 seconds.
starting >>> catkin_tools_prebuild
Finished<<< catkin_tools_prebuild
starting >>> mavlink
Finished<<< mavlink
starting >>> mavros_msgs
Finished<<< mavros_msgs
starting >>> libmavconn
Finished<<< libmavconn
starting >>> mavros
[build -07:36.6] [4/7 complete] [1/1 jobs] [0 queud] [High Load] [mavros:make (2%) -41.8]

^this is the moment is crashes, it has crashed twice in the same spot of make (2%)

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2018-02-09:
Did you installed additional heatsink (and maybe cooler fan)?
Mavlink and msgs packages are relatively light load (runs python generators), while libmavconn and mavros does lots of C++ coge compillation.

chngdickson, 2018-02-09:
@vooon i haven't... It's on it's way atm. However, doesn't -j1 and -l1 solve the problem of thermal throttling by only utilizing 25% of the CPU capacity. Also, it seems to always fail at mavros around 2% installation.
My other question would be... based on ur vast experience, do u think it's an installation issue or just a heat issue?
Thank you for ur quick response.

vooon, 2018-02-09:
I suggest overheat because you says that pi stop responding. Other possible thing is faulty power supply.
You may cd to mavros/mavros and `catkin build --this -iv -j1`, that command should show all CMake messages.

chngdickson, 2018-02-11:
Solved, Thanks @vooon

closed 2018-02-11 by chngdickson

ryanpalski, 2019-07-19:
@chngdickson was it indeed the overheating or did you see anything when building verbose? Having the same exact issue.

chngdickson, 2019-07-19:
@ryanpalski it was overheating. It was solved after i put a literal fan right next to it.

ryanpalski, 2019-07-19:
@chngdickson Sweet thanks.