mavlink/mavros #945: My ground rover don't move with setpoint_velocity/cmd_vel

nanane3437, 2018-02-15:
----
### Issue details
Hi,

I try to move my ground robot with **/mavros/setpoint_velocity/cmd_vel** but when I used it nothing happen.
I try to follow this tutorial : https://pixhawk.org/dev/ros/ground_rover 
I have : 
 - a ROS Kinectic
 - an Odroid xu4 on Ubuntu Mate 
 - mission planner
 - a pixhawk connected to Odroid by FTDI on TELEM2
 - a ground rover

So, step by step : 
> $ roscore

I run a mavproxy and a mavros

> $ sudo mavproxy.py --master=/dev/ttyUSB0 --baudrate 115200 --out udp:127.0.0.1:14540  --out udp:127.0.0.1:14550 --console

> $ roslaunch mavros apm.launch fcu_url:=udp://:14550@

Then I rosrun send_to_cmdvel.cpp in my rover_test package

> $ rosrun rover_test send_to_cmdvel

my code send_to_cmdvel.cpp : 

```
#include <ros/ros.h>
#include <geometry_msgs/TwistStamped.h>

int main(int argc, char *argv[])
{
    ros::init(argc, argv, "cmd_vel_fusion");
    ros::NodeHandle nh;
    ros::Publisher send_velocity_pub = nh.advertise<geometry_msgs::TwistStamped>("/mavros/setpoint_velocity/cmd_vel", 1000);
    ros::Rate loop_rate(100);

    geometry_msgs::TwistStamped send_velocity_msg;

    double ros_roll=0.0;
    double ros_pitch=0.0;
    double ros_yaw=0.0;
    double ros_throttle=0.0;
    int count = 1;

    while (ros::ok())
    {
        nh.param<double>("ros_roll", ros_roll, 0.0);
        nh.param<double>("ros_pitch", ros_pitch, 0.0);
        nh.param<double>("ros_yaw", ros_yaw, 0.0);
        nh.param<double>("ros_throttle", ros_throttle,0.0);

        send_velocity_msg.header.stamp = ros::Time::now();
        send_velocity_msg.header.seq = count ;
        send_velocity_msg.header.frame_id = 1 ;

        send_velocity_msg.twist.linear.x = ros_throttle;
        send_velocity_msg.twist.linear.y = 0.0;
        send_velocity_msg.twist.linear.z = 0.0;
        send_velocity_msg.twist.angular.x = ros_pitch;
        send_velocity_msg.twist.angular.y = ros_roll;
        send_velocity_msg.twist.angular.z = ros_yaw;

        send_velocity_pub.publish(send_velocity_msg);
        ros::spinOnce();
        count++;
        loop_rate.sleep();
    }
[... 467 lines cut ...]
      - 
        key: "Voltage"
        value: "12.75"
      - 
        key: "Current"
        value: "19.1"
      - 
        key: "Remaining"
        value: "9.0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1358 messages, from 2 addresses
sys:comp   list of messages
 255:190   0
  1:1     0, 129, 2, 22, 1, 152, 150, 24, 27, 30, 193, 33, 35, 36, 165, 42, 173, 62, 65, 74, 77, 163, 241, 116, 125

```

TSC21, 2018-02-17:
@nanane3437 https://pixhawk.org/dev/ros/ground_rover is probably deprecated already. You should follow https://docs.px4.io/en/frames_rover/ together with https://dev.px4.io/en/ros/mavros_offboard.html. Though I am currently not sure what is the current support of the rover type for OFFBOARD control.

TSC21, 2018-02-17:
Oh wait. Now I see you are using APM. Forget the previous post. You should also set it to GUIDED mode if you want the vehicle to respond at all. https://pixhawk.org/dev/ros/ground_rover is for PX4 Pro autopilots only. For rovers probably you will find better info in http://ardupilot.org/rover/.

nanane3437, 2018-02-17:
In http://ardupilot.org/rover/ there is nothing about using the rover with mavros ... :/

TSC21, 2018-02-17:
> In http://ardupilot.org/rover/ there is nothing about using the rover with mavros ... :/

Well then unless someone from APM can help you out, you won't find much of a help here unless you dig in yourself. I suppose APM Rover supports GUIDED mode, so from there you should be able to control the vehicle. But I can't say much more about it since I am not an APM user/dev.

closed 2018-05-04 by TSC21