mavlink/mavros #946: Set Target Position Issue with APM

mellonUAV, 2018-02-16:
Hello,

I am trying to control a quadrotor indoors using UWB to localize. Basically UWB gives me x, y, z position data. I have managed to feed this position data as GPS to APM succesfully. Here is my previous issue on this topic:
https://github.com/mavlink/mavros/issues/942

Now I am trying to send target position to APM through mavros. I tried sending through /setpoint_raw/global. I published to this topic and set the quadrotor GUIDED mode. APM Planner didn't seem to receive any position target data. As I said in the previous issue, I am new to this stuff. Am I doing something wrong here or is there a better method to do this?

Thank you so much for your help!

Best regards,
Melih
### MAVROS version and platform

Mavros: 0.18.3 (Erle Brain-3) (I am not sure how to properly check this. They say this is the generic version in Erle Brain but I am not sure. I installed mavros to PC by "sudo apt-get ...")
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: V3.5.0-rc9 (242c8f0b)

### Node logs

```
~/erle_ws$ roslaunch mavros apm.launch fcu_url:=udp://10.0.0.2:6001@ gcs_url:=tcp-l://
... logging to /home/melih/.ros/log/1d52e220-0bed-11e8-9670-d017c21fd173/roslaunch-melih-N552VW-11985.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://melih-N552VW:35401/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
[... 364 lines cut ...]
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: udp://10.0.0.2:6001@
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
```

### Check ID

```
rosrun mavros checkid
OK. I got messages from 1:1.
```

TSC21, 2018-02-17:
@khancyr maybe you can lend a hand?

mellonUAV, 2018-02-19:
@TSC21 @khancyr 
Are these settings in the apm_config.yaml correct for what I am trying to do?

Also here is my ros node that publishes to /setpoint_raw/global:

`// code for sending target GPS position
#include <ros/ros.h>
#include <cstdlib>
#include <math.h>
#include <stdlib.h>
#include <stdio.h>
#include <mavros_msgs/GlobalPositionTarget.h>
#include <geometry_msgs/PoseStamped.h>

int main(int argc, char **argv) {

  ros::init(argc, argv, "setpoint_global");
  ros::NodeHandle n;

  ros::Publisher pub = n.advertise<mavros_msgs::GlobalPositionTarget>("setpoint_raw/global", 100);
  
  ros::Rate rate(10);

  while(ros::ok()){

	// mavros_msgs::GlobalPositionTarget msg;
    geometry_msgs::PoseStamped msg;

    msg.header.stamp=ros::Time::now();

    float longi = double(atoi(argv[1]));
    float lat = double(atoi(argv[2]));
    float alt = double(atoi(argv[3]));
    
    // msg.longitude = longi;
    // msg.latitude = lat;
    // msg.altitude = alt;

    msg.pose.position.y = longi;
    msg.pose.position.x = lat;
    msg.pose.position.z = alt;
    ROS_INFO("Position data sent %f, %f, %f", msg.pose.position.x, msg.pose.position.y, msg.pose.position.z);
	pub.publish(msg);
	// rc_override_pub.publish(msg_override);
    ros::spinOnce();
	rate.sleep();
	}
}`

TSC21, 2018-05-05:
@mellonUAV is this still a problem?