mavlink/mavros #948: Simple example for setpoint's

Ahrovan, 2018-02-16:
Is a simple example for the following to get the right concept to get started:
1. setpoint_accel
2. setpoint_attitude
3. setpoint_position
4. setpoint_raw
5. setpoint_velocity

### Autopilot Support
APM, PX4

TSC21, 2018-02-17:
@Ahrovan what is the purpose of opening this issue?

Ahrovan, 2018-02-20:
This issues has been opened to promote the concept of any of the above. But surely here is not the right place to bring an example and discuss how to use them. Please close this issues and introduce the best place to discuss how to use them.
Thanks

TSC21, 2018-02-20:
There are examples for the above on the devguides of the autopilots. At least for PX4. Don't know if there are any for APM. Though this would be autopilot specific cause it depends on the flight mode (OFFBOARD or GUIDED) and on the support for each type of setpoint. Though this should be moved to the respective autopilots discussion forums. Closing.

closed 2018-02-20 by TSC21