mavlink/mavros #951: sending `obstacle_distance` brakes odom covariance

ChristophTobler, 2018-02-19:
If I use the `odom` plugin, it works as expected. However, if I use the `obstacle_distance` plugin at the same time it "brakes" the `odom` [covariance](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/odom.cpp#L155) (part of the covariance are either `nan` or `inf`).

I searched for the bug for quite some time now and here are some things I found:
* It doesn't happen if I comment [this line](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/obstacle_distance.cpp#L82)
* It still happens if I comment https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/obstacle_distance.cpp#L67-L81
* I doesn't happen if I remove https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L4538 or I change `distances[72]` to `distance` (with according changes in the plugin)

Everything has been tested with latest mavros/mavlink master

TSC21, 2018-02-19:
The size of data being sent probably is swamping the serial link. Have you tried to reduce the size of the obstacle_distance to half?

ChristophTobler, 2018-02-19:
Fix in https://github.com/mavlink/mavros/pull/952. It was using an uninitialized matrix which seems to use the same memory as the `obstacle_distance` plugin.

TSC21, 2018-02-19:
Also what is the baud rate of your link?

closed 2018-02-19 by TSC21