mavlink/mavros #957: Mission upload broken on mavros 0.23.0 with PX4

ndepal, 2018-02-26:
On Mavros 0.23.0 uploading a mission to PX4 via the waypoint push service is broken.
PX4 (SITL) rejects the mission waypoints with the warning
```
WARN [mavlink] WPM: Unexpected mission type (112 0)
```

Mavros retries the upload a few times an then fails.

Adding the line
```
ret.mission_type = (uint8_t)mavlink::common::MAV_MISSION_TYPE::MISSION;
```
below https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/waypoint.cpp#L106 fixes the issue.

### MAVROS version and platform

Mavros: 0.23.0, default `px4.launch`
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version
PX4
Version: 1.7.3

vooon, 2018-02-26:
Problem is here because in Mavlink v2 [MISSION_ITEM][1] now have extension field, that should be filled with some value.
I suppose that for v1 PX4 fills with TYPE::MISSION.
And i'm not fully sure that that is not PX4 bug, because for extended fields you **should** expect that it might be zero in software that know nothing about that ext.

[1]: https://mavlink.io/en/messages/common.html#MISSION_ITEM

ndepal, 2018-02-26:
The problem isn't that it's zero when it shouldn't be. It's 112 when it should be 0.

LorenzMeier, 2018-02-26:
@vooon Sounds like there are missing field initializers / no default constructor? That will likely remain an issue whenever we add fields to messages.

vooon, 2018-02-26:
@LorenzMeier if i remember correctly in C++11 "Class Instance{};" should be zeroed.

Message classes do not have initialisers, i wanted to use "initialiser list" feature of C++11,
but it do not work because class should not have virtual members...

vooon, 2018-02-26:
And of-course C-C++ interop unit test do not have case to that situation.

eric1221bday, 2018-02-27:
I can confirm that @ndepal fix does resolve the issue, I think his fix should be sufficient for this situation? Since the plugin is explicitly to push mission waypoints, unless we want to generalize to pushing geofence and rally points, which are the two other types.

commit 415b789e (2018-02-27): plugin fix #957: set MISSION_ITEM::mission_type