mavlink/mavros #958: Unable to connect to the FCU with my APM

onyezem96, 2018-02-26:
I have been unable to run the offboard code found here for my APM. I know that APMs and PX4 have different modes so I have changed some of the code to account for that but still no success. I think the apm still cannot connect to the FCU connection. Also pardon my formatting, this is my first time creating an issue and i am a novice programmer.
### MAVROS version and platform

Mavros: i'm not sure
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: ?3.7.1?i'm not sure

### Node logs

```
started roslaunch server http://michael-HP-ENVY-x360-Convertible:40749/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:57600
 * /mavros/gcs_url: udp://:14560@127....
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
[... 275 lines cut ...]
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---

```

### Check ID

```
ERROR. I got 2 addresses, but not your target 1:1

---
Received 3 messages, from 2 addresses
sys:comp   list of messages
  0:8     8
 132:8     40, 8

```

vooon, 2018-02-26:
Did you able to connect with QGC/APM Planner via /dev/ttyACM0?
Maybe it need some setup.

onyezem96, 2018-02-26:
Yes I did. When I try to run both however I get a new error saying 

[FATAL] [1519661387.212448533]: FCU: DeviceError:serial:open: Device or resource busy
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/michael/.ros/log/76baae2a-1b0f-11e8-91a5-d0577b40b3a2/mavros-2*.log
Initiating shutdown!
================================================================================

TSC21, 2018-02-26:
You cannot connect them at the same time. If the serial is open to a port, cannot be reserved accessed by another,

onyezem96, 2018-03-07:
It helped a bit. It made me realize that my baud rate was set too low. It worked with 115200 and not 57600 for some reason.

closed 2018-03-07 by onyezem96