mavlink/mavros #959: PR: request param: PX4 +MAVROS

Ahrovan, 2018-02-28:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I can't read mavros/global_position/global and other values real time. 

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04
[X] PX4 (last)

### Node logs

```
roslaunch mavros px4.launch fcu_url:="serial:///dev/ttyUSB0:57600"
... logging to /home/cv/.ros/log/6a2de8ae-1c75-11e8-860a-ede8e7dd15d9/roslaunch-ubuntu-17733.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:44739/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
[... 517 lines cut ...]
        key: "Last dt (ms)"
        value: "-9.916274"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "4590.244069000"
      - 
        key: "Time offset (s)"
        value: "1519810930.376954794"
```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 433 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 36, 230, 33, 74, 231, 141, 111, 241, 242, 147, 245, 22, 24, 30
```

vooon, 2018-02-28:
Looks like you have problems with serial link quality (do you use shielding?).
And perhaps 57k is too slow for companion link. Try sub 1M (but electrical interference may become more trouble).

Ahrovan, 2018-02-28:
I have three questions @vooon :

1. What is the correct choice for [?ids=sysid,compid] in PX4 for MAVROS command.
2. In my case TELEM2 connected to MAVROS in Companion Computer with 57600 baud rate (checked with PX4 SYS_COMPANION parameter), And TELEM1 connected to QGC with XBee-Pro® 900HP (XBP9B-DMST-002), but I can't find any PX4 parameter for TELEM1 setting.
3. warning "TM: Clock skew detected (-0.050610138 s). Hard syncing clocks." says Companion Computer has (-0.050610138 s) difference with what?! and how to solve that?

vooon, 2018-03-01:
1. Leave defaults. This arg overrides mavros ids.
2. Check SYS_COMPANION docs, it has several special values for things like OSD. Try 921600.
3. Is result of low link speed or wrong configuration. Right now may be ignored.

Ahrovan, 2018-03-01:
1. I had to change it when I used apm. Now that I use px4, I said maybe it should change
2. [dev.px4.io](https://dev.px4.io/en/companion_computer/pixhawk_companion.html):

> 
0 to disable MAVLink output on TELEM2 (default)
921600 to enable MAVLink output at 921600 baud, 8N1 (recommended)
57600 to enable MAVLink output at 57600 baud, 8N1
157600 to enable MAVLink in OSD mode at 57600 baud
257600 to enable MAVLink in listen-only mode at 57600 baud

> 
My question is all that is said, is just about TELEM2? How about TELEM1 and where is parameters for TELEM1 setting?
3. Wrong configuration in what case or setting? In PX4 or MAVROS

lamping7, 2018-03-01:
Just set SYS_COMPANION to 921600 and try `roslaunch mavros px4.launch fcu_url:="/dev/ttyUSB0:921600"`. This has nothing to do with TELEM1.

Ahrovan, 2018-03-01:
@lamping7 ok I khow that. now how to change setting for TELEM1 and other serial port's

closed 2018-05-04 by TSC21

alexwb1, 2018-05-07:
Please please help me I have th exact same problem!!!

alexwb1, 2018-05-07:
My research project still needs to get off the ground, and I can make it work. I have the same exact problem and I am so frustrated by this error... Has anyone figure this out?? Please help and post what you did to make it work

Ahrovan, 2018-05-07:
@alexwb1 Just set SYS_COMPANION to 921600, I try this and problem solved

hamdaan19, 2022-05-10:
> @alexwb1 Just set SYS_COMPANION to 921600, I try this and problem solved

How do I set SYS_COMPANION to 921600. Can you please mention the steps to do so? @Ahrovan