mavlink/mavros #961: using fake_gps plugin and vision_pose_estimation plugin with Ardupilot Firmware

SubMishMar, 2018-03-02:
### Issue details

I want to control my drone ([SkyViper](http://sky-viper.com/product/info/V2450GPS-sd)) indoors by using the onboard camera for pose estimation. Actually I am looking a scene having lots of ArUco markers and estimating my drone's pose by using opencv's aruco detection and pose estimation algorithm. Long story short, I am using vision to estimate the drones 6-D pose. Now I intend to send this data to the FCU for controlling the drones in guided mode indoors. There are apparently two things that I can do:
a. use fake_pgs plugin.
b. vision_pose_estimate plugin
But I am running into problems with both. I want to describe what I have done for both of the aforementioned approaches so that the reader can help me out incase I went wrong somewhere.

a. using fake_gps
1. Added fake_gps plugin to my apm_config.yaml.
2. Edited apm_pluginlists.yaml and here is how the fake_gps part looks:

```
# fake_gps
fake_gps:
  # select data source
  use_mocap: true       # ~mocap/pose
  mocap_transform: true   # ~mocap/tf instead of pose
  use_vision: false       # ~vision (pose)
  # origin (default: Zürich)
  geo_origin:
    lat: 30.6197674          # latitude [degrees]
    lon:  -96.3414215          # longitude [degrees]
    alt: 99.9674829152          # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 2.0
  epv: 2.0
  satellites_visible: 10   # virtual number of visible satellites
  fix_type: 3             # type of GPS fix (default: 3D)
  tf:
    listen: false
    send: false           # send TF?
    frame_id: "map"       # TF frame_id
    child_frame_id: "Quadrotor_pose" # TF child_frame_id
    rate_limit: 10.0      # TF rate
  gps_rate: 5.0           # GPS data publishing rate
```
3. I am publishing the robot pose over the rostopic /mavros/fake_gps/mocap/tf at about 4 Hz (I can increase it upto 30 Hz but I think the gps_rate parameter limits it, btw can I increase gps_rate?)
4. I echo  /mavros/global_position/global  I dont see any meaningful outputs,

Can anyone please help me here to get this thing working? Is there anything else I need to do? Any parameters that I must set or unset?

b. using vision_pose
 
1. Added fake_gps plugin to my apm_config.yaml.
2. Edited apm_pluginlists.yaml and here is how vision_pose part looks:

```
# vision_pose_estimate
vision_pose:
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "map"
    child_frame_id: "vision_estimate"
    rate_limit: 10.0
```
3. I publish the robot pose over the rostopic /mavros/vision_pose/pose
4. I make some modifications in the vision_position_estimate.cpp  and use `VISION_POSITION_DELTA `instead of `VISION_POSITION_ESTIMATE`.  
5. I set appropriate flags in the above code to see if this code is executed and those flags are printed, so i conclude that mavros is sending data to FCU but there is probably no reception on the FCU end.

Lot of time spent on this, any help is much appreciated!!
P.S. I was able to do the vision_position_estimate in PX4 firmware, havent tried fake_gps there but due to some constraints, I want to make this work on arducopter firmware.

### MAVROS version and platform

Mavros: 0.18
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[... 387 lines cut ...]
    values: 
      - 
        key: "Voltage"
        value: "0.00"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1131 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 193, 152, 150, 24, 27, 29, 30, 33, 163, 36, 165, 42, 178, 182, 62, 65, 74, 77, 241, 116, 125

```

mellonUAV, 2018-03-06:
Hello @SubMishMar,

I was also trying to use fake_gps plugin but I wasn't able to figure out how to use it but here is what I did, maybe it will help you:

I created a ros node that takes position data from Mocap system (UWB in my case) and converts it to latitude longitude values. Then I published these values to /mavros/hil/gps topic. Then I changed the GPS_TYPE parameter in mavros to 14. I was able feed the position data to some degree but it was not successful as I wanted, I have some problems in receiving a good position hold. I am trying to adjust values to get a better result. Hope this helps. I would appreciate if someone explains how to properly use fake gps plugin as well :)

Here is the ros node code:

 `// adding x, y, z to long lat transform
// v3 - taking data from UWB and sending to APM

#include <ros/ros.h>
//#include <geometry_msgs/PoseStamped.h>
#include <cstdlib>
#include <mavros_msgs/HilGPS.h>
//#include <mavros_msgs/OverrideRCIn.h>
#include <math.h> 
#include <stdlib.h>
#include <stdio.h>
#include <array>
#include <geometry_msgs/PoseStamped.h>

std::vector <float> GPS = {0, 0}; // {long, lat}
float xPos = 0;
float yPos = 0;
float zPos = 0;

std::vector <float> CartToGPS (float x, float y, float z){ // x, y, z in [mm]

  std::vector <float> GPS = {0, 0}; // {long, lat}

  //Some constants
  float pi = 3.14159265359; // pi
  float alpha = pi / 2; // angle between x axis and North
  float r = 6371000000; // mean radius of Earh in [mm]
  float lat0 = (39.894470 * pi) / 180; // origin latitude
  float long0 = (32.777973 * pi) / 180; // origin longitude

  float dlat = (x * cos(alpha) + y * sin(alpha)) / r; // [rad]
  float dlong = (x * sin(alpha) - y * cos(alpha)) / r; // [rad]

  GPS = {((long0 + dlong) * 180) / pi, ((lat0 + dlat) * 180) / pi};

  return GPS;
}

void poseMessageReceived(const geometry_msgs::PoseStamped &msg) // callback function for pose Message
{
  xPos = msg.pose.position.x;
  yPos = msg.pose.position.y;
  zPos = msg.pose.position.z;
  //ROS_INFO_STREAM("callback");
}

int main(int argc, char **argv) {

  // if(argc<4){
  //   printf("Usage: rosrun send_pose position_sender lat long alt\n");
  //   return -1;
  // }

	//int rate = 10;

  ros::init(argc, argv, "position_sender");
  ros::NodeHandle n;

  ros::Publisher pub = n.advertise<mavros_msgs::HilGPS>("mavros/hil/gps", 1);
  ros::Subscriber subPose = n.subscribe("UWB_Pos",5,&poseMessageReceived);
    
  ros::Rate rate(5);

  while(ros::ok()){

		mavros_msgs::HilGPS msg;

		//msg.header.frame_id = "1";
    msg.header.stamp=ros::Time::now();

    // float x = double(atoi(argv[1]));
    // float y = double(atoi(argv[2]));
    // float z = double(atoi(argv[3]));

    GPS = CartToGPS(xPos, yPos, zPos);

    //msg.header.stamp=ros::Time::now();

    msg.fix_type = 3;
    
    msg.geo.longitude = GPS[0];
    msg.geo.latitude = GPS[1];
    msg.geo.altitude = zPos / 10000;

		// msg.eph = 65535;
		// msg.epv = 65535;
		// msg.vel = 65535;
  //   msg.vn = 0;
  //   msg.ve = 0;
  //   msg.vd = 0;
  //   msg.cog = 65535;
  //   msg.satellites_visible = 255;

    msg.eph = 2.0;
    msg.epv = 2.0;
    msg.vel = 65535;
    msg.vn = 0;
    msg.ve = 0;
    msg.vd = 0;
    msg.cog = 65535;
    msg.satellites_visible = 255;

    //ROS_INFO("Position data sent %f, %f, %f", msg.geo.latitude, msg.geo.longitude, msg.geo.altitude);

		pub.publish(msg);
		// rc_override_pub.publish(msg_override);

    ros::spinOnce();
		rate.sleep();
	}	
}`

SubMishMar, 2018-03-06:
@mellonUAV  thanks.. i am going to try this and will let you know how it goes.

TSC21, 2018-05-04:
@mellonUAV for `fake_gps` make sure that APM processes `HIL_GPS` msgs by default without the need of setting any parameter or so. Or maybe you just have to set `GPS_TYPE` to `14` as @SubMishMar said. For `VISION_POSITION_DELTA` I cannot say much, it really depends on the autopilot end. @khancyr any recommendations here?
@SubMishMar your code is appreciated but `fake_gps` does exactly the conversion you need. What don't you understand on its usage?

YiefengYang, 2018-05-27:
Hi! @TSC21 
      I want our PX4 to work in mode OFFBOARD or POSCTL, both are acceptable. We have mocap system OptiTrack. I have read many issues on github about px4, mavros and mocap. 

But I still have some questions:

1. Many users have mentioned the ros package 'mavros', but I want to know where this 'mavros' came from?? Does it comes from an installation command "sudo apt-get install ros-kinetic-mavros" or comes from git clone? Is there any difference between the two? At first, I didn't use the git one, after "apt-get install", "roslaunch mavros px4.launch"  can work. But last week I got the source code by git clone, and in this source code, there also exists a "px4.launch"!!
So, I want to know while we using mavros to control our PX4(Set_Point_Position, Set_Point_Attitude and so on......), which 'mavros' should we use??
2. I also see some users mentioned "mocap_pose_estimate" plugin. Many users said that this plugin can transform the data from mocap to the coordinate system that fix the PX4, but I don't know how to use it.what it is?  Is it A node? A topic like '/mavros/mocap/pose'? Or just some functions that we can use directly?
So, I want to know what is it and how to use it? How can I add this "plugin" to my own project?? 

Thank you very much!
Best wishes!!