mavlink/mavros #963: Fail to install Geographiclib dataset

Changliu52, 2018-03-03:
Hi,

I have had a couple of tries to reinstall mavros from source today (mavlink 2018.2.2-0, mavros 0.23.0).
However, when I run `install_geographiclib_datasets.sh`, it stated installing the datasets but only gives emply folder `/usr/local/share/GeographicLib/geoids`, `/usr/local/share/GeographicLib/gravity`, `/usr/local/share/GeographicLib/magnetic`,

Then, when I try to run `sudo geographiclib-get-geoids egm96-5`, it gives:
```
download egm96-5.tar.bz2 ...
--2018-03-03 11:58:48--  http://downloads.sourceforge.net/project/geographiclib/geoids-distrib/egm96-5.tar.bz2?use_mirror=autoselect
Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 13.32.67.159, 13.32.67.197, 13.32.67.206, ...
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|13.32.67.159|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://downloads.sourceforge.net/#!/project/geographiclib/geoids-distrib/egm96-5.tar.bz2 [following]
--2018-03-03 11:58:48--  https://downloads.sourceforge.net/
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|13.32.67.159|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 652 [text/html]
Saving to: ‘/tmp/geoid-Nj9lOhZL/egm96-5.tar.bz2’

/tmp/geoid-Nj9lOhZL 100%[===================>]     652  --.-KB/s    in 0s      

2018-03-03 11:58:49 (16.9 MB/s) - ‘/tmp/geoid-Nj9lOhZL/egm96-5.tar.bz2’ saved [652/652]

unpack egm96-5.tar.bz2 ...
bzip2: (stdin) is not a bzip2 file.
tar: Child returned status 2
tar: Error is not recoverable: exiting now
```

This looks like corrupted downloads. 
But I have tried multiple times, in different days, still the same error.
Not sure what else I can try?

Thanks in advance.
Chang

TSC21, 2018-03-03:
I actually confirm this is happening also. This seems to be happening when the file is download with `wget`. A quick way around is to copy and paste https://downloads.sourceforge.net/#!/project/geographiclib/geoids-distrib/egm96-5.tar.bz2 to your browser, which automatically downloads the file, and then extract and copy the `geoids` folder to `/usr/share/GeographicLib`. I am going to check what may be the problem here.

TSC21, 2018-03-03:
I think something may be wrong with the GeographicLib web platform, since I can't even access it - https://geographiclib.sourceforge.io/. It may be that it is down for a while and when it comes back up probably the script will work properly. @Changliu52 please follow the workaround above for now. As this is not related with MAVROS, I am closing.

closed 2018-03-03 by TSC21

Changliu52, 2018-03-03:
Awesome. Thanks for the help Nuno.

Pedro-Roque, 2018-03-05:
For me, I fixed related issues with geographiclib by installing it from source with cmake:
https://geographiclib.sourceforge.io/html/install.html#cmake

...and then installing geographiclib-tools from debian repo's . That fixed all dependencies

TSC21, 2018-03-05:
@Pedro-Roque the problem is not the dependencies but actually the download of the datasets when using the `geographiclib-tools`.

Pedro-Roque, 2018-03-05:
Should have read it slower! That might actually be their servers, just a matter of time until they fix it I guess

karthikdesai, 2022-03-04:
Haha after 4 years, i am still facing this issue. It is as the problem is described. So weird

Shahzaib210749, 2024-09-07:
i want to download egm96-5.tar.bz2 manually and have found many redirections like its say we start downloading in a while but not downloadin
[image]
[image]
g