mavlink/mavros #965: PX4Flow sensor data

TrishantRoy, 2018-03-05:
### Issue details
Hello, I am using OdroidXU4+Pixhawk for autonomous control. I am using a PX4Flow sensor attached to the I2C port of the Pixhawk. The PX4Flow is working and the data is showing in Mission Planner.
I am able to get sensor data when PX4Flow is connected via micro USB (by using `roslaunch mavros_extras px4flow.launch`).
How to get the sensor data through code when connected to Pixhawk? I tried rostopic echo from the topic: /mavros/px4flow/ground_distance but no data is being published. 
I had done `rosservice call /mavros/set_stream_rate 0 10 1` without which no mavros data was coming.

Thank you for your help in advance.

### MAVROS version and platform

Mavros: 0.22.0 (I am not sure... How to check?)
ROS: Kinetic
Ubuntu: 16.04 Mate

### Autopilot type and version
APM 3.5.5

### Node logs
```
odroid@odroid:~/mavros_ws$ roslaunch mavros apm.launch fcu_url:="/dev/ttyACM0"
... logging to /home/odroid/.ros/log/10b7e9b8-d0dd-11e5-8f72-c412f52e0e75/roslaunch-odroid-1723.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:60714/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
[... 338 lines cut ...]
    values: 
      - 
        key: "Voltage"
        value: "0.00"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

```

### Check ID

```
odroid@odroid:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 3345 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 65, 24, 150, 152, 27, 29, 30, 33, 35, 36, 165, 42, 178, 182, 62, 193, 74, 77, 163, 100, 241, 116, 125
```

TSC21, 2018-03-05:
> How to get the sensor data through code when connected to Pixhawk?

Quick answer: look into the docs - http://ardupilot.org/copter/docs/common-px4flow-overview.html.

closed 2018-03-05 by TSC21

TSC21, 2018-03-05:
To complement the above: looks like Ardupilot does not publish `OPTICAL_FLOW_RAD` msgs, but rather `OPTICAL_FLOW` (https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L1346-L1376). That means that if you want MAVROS to receive those msgs, you have to extend the [`px4flow.cpp` plugin](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/px4flow.cpp).

cpamritkar, 2019-03-11:
Hello all,
I am also facing same issue when connected to PC via USB and try to read the MAVLink packets. The sensor is communicating fine when accessed through MissionPlanner.

Any advice for the same is appreciated.