mavlink/mavros #967: Question about specifying range_finder port or otherwise rode to run.

mubaarik, 2018-03-06:
I am setting up ROS on an intel aero drone and have a range finder(teraranger one) connected via USB port. I have installed the range finder ROS package and have it running on the compute board as a node well. However, I am not sure of how to connect this node to Mavros such that Mavros uses this package and the USB port to find the range finder data for the  topic /mavros/rangefinder/rangefinder. 

Any suggestions? Thanks

TSC21, 2018-03-06:
You are looking things the wrong way. MAVROS would be listening to a USB port where sends and receives Mavlink msgs. If you want to listen to a specific usb port to get the data from a rangefinder, then you need a driver, a ROS wrapper and a node to publish that data on the ROS framework. The topic you specificied publishes rangefinder data received from the Mavlink stream, more specifically published by the Ardupilot Firmware.

mubaarik, 2018-03-06:
Nuno, 

Thank you for your response. So just to be clear, I have the ROS wrapper(https://github.com/Terabee/terarangerone-ros), so what I want is the Mavros node to listen from nodes in that wrapper instead of going to the Mavlink stream and populate the corresponding topics accordingly. Does that make sense? 

I could launch the range finder separately, but just for things to look more integrated. 

On the other case, I could set-up the flight controller such that the firmware uses this sensor with the given setup for getting the range finder data, but that seems a lot of work on a customized platform like the one I have. I don't have much experience with flight controllers and drones in general. 

So any further suggestions?

Thanks

vooon, 2018-03-07:
You can remap topics. http://wiki.ros.org/roslaunch/XML/remap

TSC21, 2018-05-04:
@mubaarik is this clarified to you?

mubaarik, 2018-05-05:
@TSC21 I ended up writing my own driver to publish mavlink messages from the compute board and sending them to the flight controller via a tcp port. I did this mainly because I needed a range sensor data on the Firmware itself for using optical-flow for position estimation. 

I was definitely not asking the right questions back then but I have learned a lot since so I am good.

 Thanks.

TSC21, 2018-05-05:
Why didn't you use the `distance_sensor` plugin then?

mubaarik, 2018-05-05:
@TSC21 I believe the Distance_sensor Plugin just streams range finder data from the Firmware, it doesn't publish anything back to the flight controller?

TSC21, 2018-05-05:
It does: https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/distance_sensor.cpp#L274

mubaarik, 2018-05-05:
Well, It would have been very helpful if you had the time to mention it back in the day. I have something that works for me now.

TSC21, 2018-05-05:
I answered your questions and concerns but I wasn't clear what you wanted to do. Good that you have a solution at least. Closing this then.

closed 2018-05-05 by TSC21