mavlink/mavros #968: Mode stuck on LOITER instead of switching to GUIDED and will not run

onyezem96, 2018-03-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

My code cannot change the mode of my UAV from LOITER to GUIDED for some reason.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:115200
 * /mavros/gcs_url: udp://:14560@127....
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
[... 396 lines cut ...]
                offb_set_mode.response.mode_sent){
		std::cout << "STATE: " << current_state.mode << std::endl;
                ROS_INFO("GUIDED enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed &&
                (ros::Time::now() - last_request > ros::Duration(5.0))){
                if( arming_client.call(arm_cmd) &&
                    arm_cmd.response.success){
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

        local_pos_pub.publish(pose);

        ros::spinOnce();
        rate.sleep();
    }

    return 0;
}
```

alvikag, 2018-03-19:
Hi,
 I have mavros 0.23.3 and ROS kinetic setup on an odroid which is communicating to 3DR solo over WiFi.  I am not able to change the mode to GUIDED it just gets stuck in LOITER or results ina time out. If I try the same commands with mavros 0.21.2, mode change works perfectly.

vooon, 2018-03-20:
@alvikag change config to set `~fcu_protocol="v1.0"`.

alvikag, 2018-03-27:
Thankyou it worked!! 
Also, I am facing another problem I am using mavros 0.23.3 and ros kinetic setup on an odroid, I am connecting a pixracer via USB. Upon launching the mavros node, I am not able to get the hearbeat connection. I even tried setting the SYS_COMPANION parameter to 921600 as recommended and also did a reboot, but it doesnt seem to be working.

closed 2018-03-27 by vooon