mavlink/mavros #969: HIL GPS message values information

mellonUAV, 2018-03-07:
Hello,

I am using HIL GPS message (/mavros/hil/gps, mavros_msgs/HilGPS) to send the position data from my positioning system. With this method I can successfully publish GPS position to my FCU. In this message, this are the areas that can ben sent:

```
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
uint8 fix_type
geographic_msgs/GeoPoint geo
  float64 latitude
  float64 longitude
  float64 altitude
uint16 eph
uint16 epv
uint16 vel
int16 vn
int16 ve
int16 vd
uint16 cog
uint8 satellites_visible
```
and these messages are sent to GPS_RAW_INT Mavlink message which has these areas:

```
time_usec | uint64_t | Timestamp (microseconds since UNIX epoch or microseconds since system boot)
-- | -- | --
fix_type | uint8_t | See the GPS_FIX_TYPE enum.
lat | int32_t | Latitude (WGS84, EGM96 ellipsoid), in degrees * 1E7
lon | int32_t | Longitude (WGS84, EGM96 ellipsoid), in degrees * 1E7
alt | int32_t | Altitude (AMSL, NOT WGS84), in meters * 1000 (positive for up). Note that virtually all GPS modules provide the AMSL altitude in addition to the WGS84 altitude.
eph | uint16_t | GPS HDOP horizontal dilution of position (unitless). If unknown, set to: UINT16_MAX
epv | uint16_t | GPS VDOP vertical dilution of position (unitless). If unknown, set to: UINT16_MAX
vel | uint16_t | GPS ground speed (m/s * 100). If unknown, set to: UINT16_MAX
cog | uint16_t | Course over ground (NOT heading, but direction of movement) in degrees * 100, 0.0..359.99 degrees. If unknown, set to: UINT16_MAX
satellites_visible | uint8_t | Number of satellites visible. If unknown, set to 255
alt_ellipsoid | int32_t | Altitude (above WGS84, EGM96 ellipsoid), in meters * 1000 (positive for up).
h_acc | uint32_t | Position uncertainty in meters * 1000 (positive for up).
v_acc | uint32_t | Altitude uncertainty in meters * 1000 (positive for up).
vel_acc | uint32_t | Speed uncertainty in meters * 1000 (positive for up).
hdg_acc | uint32_t | Heading / track uncertainty in degrees * 1e5.
```
It seems that there is no correcponding vn, ve, vd values in this messages. I couldn't find much information about this mavros_msgs/HilGPS like what are the unit types of vn, ve, vd values and how are they interpreted in the FCU. Also what should I input to "cog" value? Right now I use 65535 since it is unknown. I am using this message because I couldn't menage to use fake_gps plugin directly, also some explanation on how to use this plugin would be appreciated. Thank you!

Best regards,
Melih Pelit

TSC21, 2018-05-04:
> I couldn't find much information about this mavros_msgs/HilGPS like what are the unit types of vn, ve, vd values and how are they interpreted in the FCU. Also what should I input to "cog" value?

https://mavlink.io/en/messages/common.html#HIL_GPS

> I couldn't menage to use fake_gps plugin directly, also some explanation on how to use this plugin would be appreciated.

What can't you understand on the plugin?

TSC21, 2018-05-31:
Closing as stale

closed 2018-05-31 by TSC21