mavlink/mavros #973: [BUG] ROS kinetic & MAVROS 0.23.0 Problem

dronecrewkari, 2018-03-14:
### Issue details

We are using ```ROS Kinetic``` and ```MAVROS v0.23.0``` for gazebo px4 SITL.

when we do rostopic echo in order to monitor each mavlink topic by 

```rostopic echo mavros_msgs/Mavlink```

Length of message ID #105(https://mavlink.io/kr/messages/common.html#HIGHRES_IMU)

is supposed to be 62 length, but it varies from **61 to 62.**, **which is the problem.**

In addition, length of other messages(```mavros_msgs/Mavlink```) are not consistent with their definition.

from time to time. 

In the past, when we tested ```ROS indigo``` and ```MAVROS v0.17.5``` for gazebo px4 SITL,

we had no problem at all, and length of each mavlink message is always consistent with their definition.

### MAVROS version and platform

Mavros: 0.23.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ v ] PX4

Version: 1.7.3

vooon, 2018-03-14:
Does it send mavlink 2?

dronecrewkari, 2018-03-14:
@vooon  

It sends mavlink v3 message

vooon, 2018-03-14:
Mavlink v3 is not exist yet. What value Mavlink::magic have?

dronecrewkari, 2018-03-14:
@vooon 

```
header: 
  seq: 463
  stamp: 
    secs: 133
    nsecs:  62000000
  frame_id: ''
framing_status: 1
magic: 254
len: 62
incompat_flags: 0
compat_flags: 0
seq: 96
sysid: 1
compid: 1
msgid: 105
checksum: 24634
payload64: [129071464, 4449852998238703290, 13501161584329541006L, 13480068643353349519L, 4219899119912512951, 3202370762, 4896982051447635968, 30783137906688]
signature: []
---
header: 
  seq: 464
  stamp: 
    secs: 133
    nsecs:  62000000
  frame_id: ''
framing_status: 1
magic: 254
len: 28
incompat_flags: 0
compat_flags: 0
seq: 97
sysid: 1
compid: 1
msgid: 30
checksum: 6693
payload64: 
  - 13587087834979891247
  - 13514568930264840325
  - 4286445697480047792
  - 13480068643355710640
signature: []

```
```
magic: 254, which means uint8 MAVLINK_V10 = 254
```

I have tried mavros ```0.22.0```, and it does not have problem at all.
 
and the length of mavros is correct all the time.

Would you please let me know how to install mavros ```0.22.0``` in not src folder(not source code), but ```/opt/ros/kinetic/share as executable?``?

If we do ```sudo apt-get install ros-kinectic-mavros```, the installed version of mavros is 0.23.0 which we don't like.

vooon, 2018-03-14:
https://askubuntu.com/questions/428772/how-to-install-specific-version-of-some-package/428778

TSC21, 2018-03-14:
This is not a problem at all. The difference to 0.23 is that we, by default, use MAVLINK v2.0. You can set it back to 1.0 on your launch file.

dronecrewkari, 2018-03-14:
```
header: 
  seq: 643
  stamp: 
    secs: 258
    nsecs: 275000000
  frame_id: ''
framing_status: 1
magic: 253
len: 61
incompat_flags: 0
compat_flags: 0
seq: 0
sysid: 1
compid: 1
msgid: 105
checksum: 5997
payload64: [10838272, 4470471791896184273, 4315112263521140020, 13575039237226844464L, 13715600795138748645L, 3201975881, 4896943345202364416, 271690235904]
signature: []
---
header: 
  seq: 644
  stamp: 
    secs: 258
    nsecs: 275000000
  frame_id: ''
framing_status: 1
magic: 253
len: 28
incompat_flags: 0
compat_flags: 0
seq: 1
sysid: 1
compid: 1
msgid: 30
checksum: 16052
payload64: 
  - 13588184705202661974
  - 4593572619793631631
  - 4243887491471530941
  - 13575039239399422531
signature: []
---
header: 
  seq: 645
  stamp: 
    secs: 258
    nsecs: 275000000
  frame_id: ''
framing_status: 1
magic: 253
len: 28
incompat_flags: 0
compat_flags: 0
seq: 2
sysid: 1
compid: 1
msgid: 33
checksum: 50559
payload64: [2035717509352532566, 2096030025249858, 281470681743504, 13575039236800905216L]
signature: []
---
header: 
  seq: 646
  stamp: 
    secs: 258
    nsecs: 276000000
  frame_id: ''
framing_status: 1
magic: 253
len: 28
incompat_flags: 0
compat_flags: 0
seq: 3
sysid: 1
compid: 1
msgid: 32
checksum: 32666
payload64: 
  - 13529686782479706710
  - 13696611677434185999
  - 13540920298374900385
  - 13575039239394931871
signature: []
---
header: 
  seq: 647
  stamp: 
    secs: 258
    nsecs: 276000000
  frame_id: ''
framing_status: 1
magic: 253
len: 32
incompat_flags: 0
compat_flags: 0
seq: 4
sysid: 1
compid: 1
msgid: 31
checksum: 26891
payload64: [4556390178507205206, 13579059537067253320L, 4315112261341295591, 13575039237226844464L]
signature: []
---
header: 
  seq: 648
  stamp: 
    secs: 258
    nsecs: 280000000
  frame_id: ''
framing_status: 1
magic: 253
len: 62
incompat_flags: 0
compat_flags: 0
seq: 5
sysid: 1
compid: 1
msgid: 105
checksum: 56736
payload64: [10843264, 13628085336831468324L, 4260370049180363824, 13451504604716200280L, 13716931960131098100L, 3202622573, 4896943345202364416, 2195835584512]
signature: []
---
header: 
  seq: 649
  stamp: 
    secs: 258
    nsecs: 280000000
  frame_id: ''
framing_status: 1
magic: 253
len: 28
incompat_flags: 0
compat_flags: 0
seq: 6
sysid: 1
compid: 1
msgid: 33
checksum: 1442
payload64: [2035717509352532571, 2096030025249858, 281470681743504, 13451504604294545408L]
signature: []
---
header: 
  seq: 650
  stamp: 
    secs: 258
    nsecs: 286000000
  frame_id: ''
framing_status: 1
magic: 253
len: 62
incompat_flags: 0
compat_flags: 0
seq: 7
sysid: 1
compid: 1
msgid: 105
```

dronecrewkari, 2018-03-14:
''' kari@choi:~$ sudo apt-get install ros-kinetic-mavros version=0.22.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package version
'''

dronecrewkari, 2018-03-14:
kari@choi:~$ sudo apt-get install ros-kinetic-mavros=0.22.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '0.22.0' for 'ros-kinetic-mavros' was not found

dronecrewkari, 2018-03-14:
http://build.ros.org/view/Kbin_uX32/job/Kbin_uX32__mavros__ubuntu_xenial_i386__binary/85/

0.22.0-0xenial-20180202-075718-0800

vooon, 2018-03-14:
Gotcha!
```
magic: 253
len: 61
```

Not a bug. Closing.

closed 2018-03-14 by vooon

dronecrewkari, 2018-03-14:
@TSC21 How can we change to set it back MAVLINK v1.0 in your ROS launch file?

dronecrewkari, 2018-03-14:
@vooon 

Thank you for the help

Do you mean that message length can vary in mavros 0.23.0 with v2.0? 

If yes, why does it vary? Why not with v1.0?

If yes, how can we decode mavros_msgs/Mavlink which varies in term of length? For instance, how can we use memcopy function to decode this message?

vooon, 2018-03-14:
Yes because it is MAVLink v2.0. And usage to/from topics outside of mavros is a bad practice.
Write a plugin. It is simple.

dronecrewkari, 2018-03-15:
@vooon 

Thank you for the kind reply.

BTW, why is the reason that usage to/from topics outside of mavros is a bad practice?

Theatrically, it should be the same as using a plugin.