mavlink/mavros #976: fly lower and land safely

raywang-nx, 2018-03-16:
### Issue details
hi there:
    i got a task here, not quite difficult, but a little bit complex.
    i got a pixhawk and joule running mavros. the job of joule is to control the the quadrotor to move forward, and avoid the obstacles, which means if there are obstacles, the joule will send the instructions to make it fly up, down, left and right. after a few studying on mavros, i decide to use mavros/setpoint_velocity to make it. here i change the px4_config.xml a little. the main changes are listed below, including the heartbeat rate, safety_area p1->z and setpoint_velocity's body_frame.

# Common configuration for PX4 autopilot
#
# node:
startup_px4_usb_quirk: true

# --- system plugins ---

# sys_status & sys_time connection options
conn:
  heartbeat_rate: 10.0    # send hertbeat rate in Hertz
  timeout: 10.0          # hertbeat timeout in seconds
  timesync_rate: 10.0    # TIMESYNC rate in Hertz (feature disabled if 0.0)
  system_time_rate: 1.0  # send system time to FCU rate in Hertz (disabled if 0.0)

# safety_area
safety_area:
  p1: {x:  1.0, y:  1.0, z:  0.20}
  p2: {x: -1.0, y: -1.0, z: -1.0}

# setpoint_velocity
setpoint_velocity:
  mav_frame: BODY_NED

     and now my testing environment is the parking lot on the b3 floor. so the height is limited to about 2.5m. considering it will fly upward to about 2m or so to avoid the obstacle, so i just want it to fly as low as possiable so i change the safety_area. in order to use the body_frame as the coordinate, i change the setpoint_velocity to BODY_NED. 
    the steps i used is that firstly, use the controller to make the drone just fly off the ground, and then change to offboard mode to let the mavros and joule take control, it suppose to fly to about 0.2 metere above the ground. but the reality is that it fly to about 1.5m and go on the algorithm. while sometimes i can control it to about 0.3m, but it is occasionally. 

1、i wanna ask that will my strategies work? or what can i do to improve the stability?
2、in order to control it to land safely, after the drone fly higher, i change the offboard mode to the land mode to land safe and sound. but it rejected to the land mode, and now i got a broken drone and will repair it latter.... before that, i will change the offboard to the stabilized mode, but in that mode i didn't give enough throttle, so it fall down immediately and ..... you know what will happen... so i wanna ask that what can i do to make it land safely from the offboard mode? (maybe i should post it in px4 again?)

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: v1.7.3

vooon, 2018-03-16:
I'm not sure that safety area actually works, at time of addition that message not supported by any firmware.
@TSC21 comments?

raywang-nx, 2018-03-16:
@vooon yes, sometimes it works, sometimes it doesn't. and now i am trying to find out how to fly lower, what do you think of the landing problem? what kind of actions should i take?