mavlink/mavros #980: increase sensor rate SITL

aaumar, 2018-03-23:
----

### Issue details

Hi
I am developing my own fixed wing control with offboard mode. But, i use 50 Hz as my sampling time but sensor just give me about 30 Hz. I have read some way to increase the rate with SD card but in SITL you don't use any SD card. I also have tried `rosrun mavros mavsys rate --all 50` but the rate didnt increase. is there any way to increase sensor rate in SITL? thank you

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?1.7.0?

TSC21, 2018-05-04:
Check `sitl_gazebo`, more in specific the `gazebo_mavlink_interface` plugin and what parameters are used there (loaded from the SDF file). This is not a MAVROS related issue.

closed 2018-05-04 by TSC21

yc718764666, 2018-06-26:
hello aaumar @aaumar ,could you solve this problem? I have the same problem.

danielhentzen, 2018-08-03:
Hi @aaumar , @yc718764666 . I came across the same question and solved it the following way:

In `Firmware/posix-configs/SITL/init/ekf2/iris` you can set the streaming rate of Mavlink messages (used for the EKF, but also works for other messages that are not used in the EKF). For instance, to set a desired rate for LOCAL_POSITION_NED on udp port 14557:

mavlink stream -r 200 -s LOCAL_POSITION_NED -u 14557

Hope that helps.

Seanmatthews, 2019-02-27:
@danielhentzen I no longer see that directory in the master branch of PX4. It's also not clear from searching where that should be set now. Anyone have any other tips?

adk4davsys, 2020-08-11:
> @danielhentzen I no longer see that directory in the master branch of PX4. It's also not clear from searching where that should be set now. Anyone have any other tips?

Hello, I don't know if you are still working on this problem but the way I changed my gazebo SITL mavlink rate is by typing:
pxh> mavlink stream -r 200 -s LOCAL_POSITION_NED -u 14557
into the terminal that is running the ros node.

Hope this helps anyone still looking to increase the mavlink rate within gazebo SITL.