mavlink/mavros #987: Trigger cycle_time cannot be changed

Changliu52, 2018-03-31:
### Issue details

I tried to use "~/cmd/trigger_control" to control the trigger of pixracer, by calling service 
```
srv.request.cycle_time = (1000/framerate_hz);
srv.request.trigger_enable = true;
```
while setting the pixracer to `time based, on command` with `GPIO`.

This method can enable/disable camera trigger without problem, but the trigger rate is always 25 Hz regardless what I set in the cycle_time field.

### MAVROS version and platform

Mavros: 0.23.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: 1.7.3

mhkabir, 2018-03-31:
The mavlink spec changed. That field is now used to reset the trigger sequence. https://mavlink.io/en/messages/common.html#MAV_CMD_DO_TRIGGER_CONTROL. 

You should set the cycle time using the PX4 parameter `TRIG_INTERVAL`, since we decided that runtime FPS change was not a very useful feature.

I should also change the mavros implementation to reflect the same. Sorry about the confusion!

Changliu52, 2018-03-31:
Aha. I see. Thank you for the quick info, Kabir.

mhkabir, 2018-03-31:
Keeping open to track mavros change of spec.

m0oz, 2018-10-11:
@mhkabir 
Did you already change the service callback according to the current mavlink cmd spec?
I could not see it in master so I fixed it and created two seperate services.
One to send DO_TRIGGER_CONTROL and another one to send DO_SET_CAM_TRIGG_INTERVAL

I would create a PR but only if this is not already incoming

mhkabir, 2018-10-11:
Yes, please do! I dropped the ball on this one, but happy to merge the fix.

mhkabir, 2018-10-11:
This is needed to fix the API of :
https://github.com/ethz-asl/realsense/tree/feature/timesync

m0oz, 2018-10-15:
mavlink currently hase two commands to control trigger state and trigger config.
mavros on the other hand only advertises one service (trigger_control)

I am not quite sure if I should rather send two mavlink comands on one service call or if it would be better pratice to add another mavros service (We'd have trigger_control and trigger_interval then)

**One mavros service 
that sends DO_TRIGGER_CONTROL and DO_SET_CAM_TRIGG_INTERVAL**
Pros: 
- Compatibility with ethz-asl/realsense/tree/feature/timesync

- Compatibility with mavcmd trigger_control

**Two mavros services
one that sends DO_TRIGGER_CONTROL and another one DO_SET_CAM_TRIGG_INTERVAL**
Pros:

- Consistency with the other mavros services (all of them send only one mavlink cmd)

- Less prone to unexpected behavior: trigger interval and integration time can only be set while triggering is disabled anyways. So having two seperate services makes more sense from a usability perspective. Compare this to the one-service solution where users who try to set trigger_enable and interval_time at once will observe unexpected behavior

mhkabir, 2018-10-15:
Please make 2 separate services, and add a documentation note that setting integration time is only possible before triggering is enabled.

m0oz, 2018-10-22:
PR #1103 
thx for reviewing
By adding a documentation note I assume you mean to add it to the [ROS wiki page](http://wiki.ros.org/mavros)?

m0oz, 2018-10-22:
thx for review
#1103 
By adding a documentation note I assume you mean to put it [here](http://wiki.ros.org/mavros)?

closed 2018-11-06 by TSC21

SasaKuruppuarachchi, 2022-01-11:
Fixed my issue. Thanks

PR #1103 (2018-11-06): Fix of trigger setup and control via mavlink cmd interface 
Adapt trigger_control service to current mavlink cmd spec. Add a new service to change trigger interval and integration time.

Fixes #987
@mhkabir