mavlink/mavros #989: RC-Like Responsiveness with Setpoint Velocity

james-mchugh, 2018-04-03:
### Issue details

Hello,
I have been working on an interesting project using Mavros which requires me to take manual control of the drone in offboard mode. I know this seems a bit counterintuitive, but I will give my reasoning for doing so. I am working on an imitation learning algorithm for a quadcopter so that the quadcopter can autonomously fly the way that a pilot would fly it. To do this, I have to be able to record the commands that a pilot gives the quad through RC, translate that to some setpoint command (I am currently using velocity), and then give that command to Mavros (currently being done through setpoint_velocity).  I have been able to implement this to manually control the drone in offboard, but it is not too responsive. It takes a noticeable amount of time to respond to the setpoint_velocity command and sometimes it does not seem to properly respond at all. This makes it hard to use this approach for obstacle avoidance and similar objectives. I imagine that there is probably a better way of doing this, but since I am new to Mavros, I am not sure what to do. Does anyone have any suggestions? To summarize, I am trying to translate manual commands through RC to a setpoint command to be used in offboard mode.

### MAVROS version and platform

Mavros: 0.22.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.7.3

TSC21, 2018-05-04:
> It takes a noticeable amount of time to respond to the setpoint_velocity command and sometimes it does not seem to properly respond at all. This makes it hard to use this approach for obstacle avoidance and similar objectives.

1. How much time are you talking about?
2. At what rates are you publishing your commands to `setpoint_velocity`?
3. What is the baud rate of your link between the unit that sends the commands and the autopilot?
4. Why don't you use `manual_control` or `actuator_control` instead?

TSC21, 2018-05-31:
Closing as stale

closed 2018-05-31 by TSC21