mavlink/mavros #993: Failed to build mavros (0.24.0)

sabbaghp, 2018-04-06:
Current Error:

........./mavros/mavros_extras/src/plugins/vision_pose_estimate.cpp:128:46: error: ‘struct mavlink::common::msg::VISION_POSITION_ESTIMATE’ has no member named ‘covariance’
   ftf::covariance_urt_to_mavlink(cov_map, vp.covariance);

New release was pushed yesterday 4/6/18.

Mavros: 0.24.0
Ros: Kinetic
Ubuntu: 16.04
Autopilot: PX4

Standard libs (geographiclib-tools, mavlink, libgeographic-dev, ros-kinetic-control-toolbox, ros-kinetic-mavros-msgs) and scripts (install_geographiclib_datasets.sh).

Is there a new _mavlink_ library not available yet?

vooon, 2018-04-06:
Did you update mavlink?

MattMgn, 2018-04-07:
Same problem here
I have first installed mavros from binaries without problem (so mavlink should be up to date, right ?). But when i compile from source, I also get this error.
Any idea ?

mhkabir, 2018-04-07:
Mavlink from apt isn't updated yet, I guess. Please also compile mavlink-gbp-release from source.

vooon, 2018-04-09:
Simple rule: if you build mavros from source then build mavlink also. Just follow [manual][1].

[1]: https://github.com/mavlink/mavros/tree/master/mavros#source-installation

sabbaghp, 2018-04-12:
Good point and rule to follow.
It all checks out. 
Thanks @vooon @mhkabir.

closed 2018-04-12 by mhkabir