mavlink/mavros #998: Using mavros to connect with pixracer over wifi via udp

jannsta1, 2018-04-11:
### Issue details

I am trying to connect to a pixracer unit with mavros. Previously I have used a companion computer that connects to pixhawk via a physical serial cable. Now I wish to use a wifi connection to communicate directly with pixracer via the ESP8266 board. I have configured the wifi which runs in station mode and happily connects with qGroundcontrol and my router (so I think the pixracer/ESP8266 are functioning and setup correctly) but I'm struggling to connect through mavros to the fcu. 

The udp address I'm using is: ```udp://:14557@192.168.1.9:14540```  - where 192.168.1.9 is the ip address of the pixracer

Does this format look OK? Perhaps there are some parameters I need to set too? I've set ```SYS_COMPANION``` to ```ESP8266 (921600 baud, 8N1)```. 

### MAVROS version and platform

Mavros: 0.22.0
ROS: Kinetic
Ubuntu: 16.04
### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.6.5

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
header: 
  seq: 15
  stamp: 
    secs: 1523479866
    nsecs: 896584974
  frame_id: ''
status: 
  - 
    level: 1
    name: "UAV_1/mavros: FCU connection"
    message: "not connected"
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Received packets:"
        value: "0"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "0"
      - 
        key: "Tx sequence number:"
        value: "141"
      - 
        key: "Rx total bytes:"
        value: "0"
      - 
        key: "Tx total bytes:"
        value: "36947"
      - 
        key: "Rx speed:"
        value: "0.000000"
      - 
        key: "Tx speed:"
        value: "9547.000000"
  - 
    level: 2
    name: "UAV_1/mavros: GPS"
    message: "No satellites"
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 2
    name: "UAV_1/mavros: Heartbeat"
    message: "No events recorded."
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Vehicle type"
        value: "Generic micro air vehicle"
      - 
        key: "Autopilot type"
        value: "Generic autopilot"
      - 
        key: "Mode"
        value: ''
      - 
        key: "System status"
        value: "Uninit"
  - 
    level: 0
    name: "UAV_1/mavros: System"
    message: "Normal"
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "UAV_1/mavros: Battery"
    message: "No data"
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 2
    name: "UAV_1/mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "udp://:14557@192.168.1.9:14540"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last dt (ms)"
        value: "0.000000"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "0.000000000"
      - 
        key: "Time offset (s)"
        value: "0.000000000"
---

```

### Check ID

```
ERROR. I got 0 addresses, but not your target 1:1
```

vooon, 2018-04-12:
What address you use for QGC?

jannsta1, 2018-04-12:
I'm using ```udp://:14556@192.168.1.5:14550``` (where 192.168.1.5 is the ip of my ground control station - was following the convention documented in https://dev.px4.io/en/simulation/ros_interface.html, even though this relates to sitl).

It's worth noting that the pixracer automatically connects to QGroundControl on startup (external to mavros) - perhaps there could be a conflict here?

vooon, 2018-04-12:
Highly likely to be. I'm not sure that ESP firmware follow that spec. Try (without QGC) leave default port, e.g. `udp://@192.168.1.5`, or even simple `udp://@`.

jannsta1, 2018-04-13:
I still can't connect with those addresses for the ground control station. I think this might require some updates to the ESP firmware but I haven't yet figured out the codebase. I will update here if there is any success with this.

TSC21, 2018-05-05:
@jannsta1 updates?

jannsta1, 2018-05-06:
I have just tried this again and managed to connect by setting the udp port values to match the default WIFI_UDP_CPORT (14555) and WIFI_UDP_HPORT (14550) parameter values (for the ESP module). Thus my fcu_url is; ```udp://:14550@<ip address of esp device>:14555```. 

I haven't done a flight test yet and a downside is that this doesn't work with QGC but I'm more than happy to the issue. Thanks for advice about the UDP addresses, it was a big help to know what area to look in.

closed 2018-05-06 by jannsta1

ryrocha, 2019-03-18:
I was able to connect with the default udp port values using that fcu_url syntax. Thank you! I figured I'd follow up with this just in case anyone else is still having the issue of getting QGC to work in conjunction with mavros.

I was able to get QGC and mavros to work by changing the cport and hport values as shown [here](https://github.com/dogmaphobic/mavesp8266/blob/master/HTTP.md) (I choose 14556 and 14551, respectively) which would make the fcu_url: `udp://:14551:@<ip address of esp device>:14556` while keeping the gcs_url: `udp-b://127.0.0.1:14555@14550`.

DEADLYBBS, 2022-06-01:
What was the latency rates for the esp8266 module? Would it be fast enough to do offboard control with motion capture data?
Asking here since this is almost the exact implementation I am using. 
I however have QGC running on a different computer, but I will be changing that shortly.

jannsta1, 2022-06-02:
I didn't end up using the pixracer all that long but from what I remember I managed to use the esp8266 to get motion capture data and it worked well for me. There were no discernable latency issues. I don't remember exactly but I think I had an onboard pc setting the control points. I think the esp might cope with this as well though - I've certainly used the built in raspberry pi wifi for both tasks a fair bit and had no issues. Just make sure you have the usual kill switches in place for when it all goes wrong :).