mavlink/mavros #1000: ROS2: Overview what deps needed to start porting

vooon, 2018-04-12:
Subj.

TSC21, 2018-04-12:
Here goes 1000 issues!!

TSC21, 2018-04-12:
Here was my start for this: https://github.com/mavlink/mavros/projects/1. Probably needs some updates.

LorenzMeier, 2018-04-16:
Hi - this is super relevant and a PR would be appreciated. What flight controller are you using with it?

TSC21, 2018-04-16:
Well while MAVROS is not ported to ROS2, the `ros1_bridge` is an acceptable solution. But to what repo are the PR's addressed to?

mikaelarguedas, 2018-04-16:
I have code changes for 3 packages ATM (more details in the gist above):
- uuid_msgs
- geographic_msgs
- mavros_msgs

AlexisTM, 2018-04-18:
Porting to ROS2, wouldn't it be the correct solution to move from MAVLink to FastRTPS for the backend communications?

TSC21, 2018-04-18:
@AlexisTM that's already an ongoing work yes.

ArkadiuszNiemiec, 2018-07-30:
Is there a branch with ROS2 code? I would like to check if I am able to help with porting and developing FastRTPS part.

TSC21, 2018-07-30:
There's no ROS2 branch code yet. It will be made public as soon it is found stable enough for it (at least for the PX4 side)

dlarue, 2018-12-24:
Looking for ROS2 also.

evandp, 2019-07-25:
Hello, what is the status of porting to ROS2? Are there plans to make the ROS2 branch public any time soon? Thanks!

TSC21, 2019-07-25:
There's no effort to port to ROS2 atm.

TSC21, 2019-07-25:
If you are interested though on the bridge between PX4 and ROS2 check https://dev.px4.io/v1.9.0/en/middleware/micrortps.html