mavlink/mavros #999: Corrupted unsorted chunks when mavros_node is down

theg4sh, 2018-04-12:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

*** Error in `/opt/ros/kinetic/lib/mavros/mavros_node': free(): corrupted unsorted chunks: 0xb0c00aa0 ***

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04 ARM (Odroid XU4)

### Autopilot type and version

[X ] PX4

Version: 1.7.3

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2018-04-12:
Please provide more data:
- Is it source install? (looks like not)
- Did you tried source install?

Unfortunately that error message do not give any usable information.

theg4sh, 2018-04-13:
> Is it source install? (looks like not)

No, it's not.

> Did you tried source install?

No, haven't tried.

It happens rarely, so, I'll try to catch that error with gdb and share a core dump for this issue.

TSC21, 2018-05-04:
@theg4sh any updates?

TSC21, 2018-05-10:
@theg4sh closing as stale. Please reopen if you still have problems with it.

closed 2018-05-10 by TSC21