mavlink/mavros #1005: Consider options for automatically providing geographiclib dataset for binary installations

tfoote, 2018-04-16:
I've been putting together a demo and was surprised by the need to manually install the geographiclib datasets even when installing from binary packages. (https://github.com/osrf/drone_demo/issues/1)

I know that there's been discussion of adding the download at build time, but that's not exactly right as building is better if it doesn't need to have internet access. And also the database is only a runtime dependency not a build time dependency, so downloading/installing it at build time doesn't make much sense.

Since this is really a runtime/packaging issue more than a build issue, I think that the best approach would be to integrate it into the packaging. Potentially as a debian post install hook. Or potentially create a debian package with the geographiclib dataset that is added as a dependency in the packaging process so that the binary installation will get the data.

This might be a package that could be submitted upstream. I'm not sure why the geographiclib debian packaging doesn't actually include the dataset itself in a separate package, or at least have an installer package that will pull it down in a post install hook.

Does either of these approaches sound preferable to others? My suggestion would be to work toward contributing the downloading tool upstream so we can just add the dependency and then it's much less patching etc.

TSC21, 2018-05-04:
> Or potentially create a debian package with the geographiclib dataset that is added as a dependency in the packaging process so that the binary installation will get the data.

I prefer this idea. Would this be something to get into `bloom` or instead a debian package to include on `apt` sources?

tfoote, 2018-05-04:
I was thinking of a separate debian package that would be resolved via rosdep. But that's less cross platform than embedding the logic in the ROS package. The post install hook could be inserted as a patch during the bloom process. And if you're not on a system where that has been inserted, you just need to run the hook manually.

TSC21, 2018-05-05:
@tfoote do you have the required bandwidth to move forward with this?

tfoote, 2018-05-07:
I'm not going to have much time to look at this, this isn't a high priority at the moment. I'll try to find a moment of downtime to look at it. But if someone wants to take a stab at it I definitely can make time to review a proposed change.

TSC21, 2018-05-08:
Let's leave it open there. If I actually get some time I will do it myself, but time is really constrained on my side currently.