mavlink/mavros #1008: handle wind_cov (px4) for wind estimation publisher

tstastny, 2018-04-17:
I'd like to handle and publish values from PX4's wind_cov msg [WIND_COV](https://mavlink.io/en/messages/common.html#WIND_COV) on mavros. The APM specific wind msg is handled currently in the vfr_hud plugin [L76](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/vfr_hud.cpp#L76). However, the PX4 specific message also includes some variances on the estimate.

I'm motivated to PR this addition -- but would appreciate your preference on how to include it (to avoid remaking the wheel a few times):
1. Should the additional variances (horizontal and vertical) on the wind estimate be put into a separate ros message type, and leave the twist as is? Or should I make a new ros msg specifically for the wind_estimation publisher, and convert the current APM handler to populate this msg?
2. Does it make sense to keep this in vfr_hud? Or rather should it be in it's own plugin?

Let me know your thoughts please. Thanks! 

@TSC21

mhkabir, 2018-04-17:
I vote for a new wind plugin!

vooon, 2018-04-18:
New plugin. Perhaps [TwistWithCovariance][1].

[1]: http://docs.ros.org/jade/api/geometry_msgs/html/msg/TwistWithCovariance.html

tstastny, 2018-04-18:
> Perhaps TwistWithCovariance.

Thanks for the recommendation. Only question here is if it is ok just to use the first two slots of the 36 length float for the two "summed" variances coming from the mavlink message (leaving the remaining 34 blank). In the future, it may be that someone updates this on the px4 side (or makes a new member to the mavlink message for full covariance). But at the moment would be quite some unused floats. 

What about keeping the twist, and then adding a [Float64MultiArray](http://docs.ros.org/api/std_msgs/html/msg/Float64MultiArray.html) for the horizontal and vertical variance?

TSC21, 2018-04-19:
@tstastny I would use [TwistWithCovariance][1] and fill up the correspondent slots of it. There's no problem on leaving the others open/empty. And yes, definitely should go to a new plugin. 

[1]: http://docs.ros.org/jade/api/geometry_msgs/html/msg/TwistWithCovariance.html

PR #1014 (2018-04-24): add wind estimation plugin
Makes a new wind estimation plugin handling both APM and PX4 specific wind messages.
- moves old APM wind publisher from vfr_hud to the new wind plugin
- the PX4 message contains a *summed* horizontal wind speed convariance (i.e. x cov + y cov). This value is placed in the first covariance array slot for now, and the vertical in the 3rd.

resolves #1008
closed 2018-04-24 by vooon