mavlink/mavros #1009: Setpoint_raw plugin do not convert data from NED to ENU for the yaw data.

AlexisTM, 2018-04-18:
### Issue details

While we can see that the transformation is called here, the transformation function might be wrong.
 
https://github.com/mavlink/mavros/blob/d7387776254b3bfee2ca0f7830d22a4de56aa3ab/mavros/src/plugins/setpoint_raw.cpp#L75-L83

* I send a `setpoint_raw/local` of -1.57. **bottom left console**
* The loopback `setpoint_raw/target_local` **right console* send me what I send (expected as it should be ENU=>NED=>ENU).
* The Mavros attitude is 0 **top left graph**
* The ATTITUDE YAW message received through MAVLink is **+**1.57. 
* The TARGET YAW message received through MAVLink is **+**1.57. 

Mavros is supposed to:
* [x] Negate the yaw 
* [ ] Rotate 90° the yaw (**missing**)

[image]

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] PX4 1.6.5

TSC21, 2018-04-18:
Solution: https://github.com/mavlink/mavros/pull/753. Can you have a go with it please?

AlexisTM, 2018-04-18:
That was my plan; I compile and test it :) 

(I am currently on simulation)

AlexisTM, 2018-04-18:
These again, the PR needs a rebase I guess

``` 
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_helpers.h:789:54: error: ‘MAVLINK_START_SIGN_STREAM’ was not declared in this scope
    MAVLINK_START_SIGN_STREAM(status->signing->link_id);
                                                      ^
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_helpers.h:791:52: error: ‘MAVLINK_END_SIGN_STREAM’ was not declared in this scope
    MAVLINK_END_SIGN_STREAM(status->signing->link_id);
```

TSC21, 2018-04-18:
Try to put mavlink_gbp_release on your catkin ws

AlexisTM, 2018-04-18:
I rebased to the current master and it builds :)

AlexisTM, 2018-04-18:
I rebased and it builds (except mavros_extras that was not needed).

AlexisTM, 2018-04-18:
#753 solves the issue of setpoint_raw.

[image]

AlexisTM, 2018-04-18:
@TSC21 Do you know a way to check the Mavros version from CPP? Any define? (To ensure to have the fix value if mavros is an older version)