mavlink/mavros #1010: CON: Got HEARTBEAT, connected. FCU: No valid autopilot

sudosurf, 2018-04-19:
### Issue details

I'm working on setting up MoCap/GCS/FCU communication with an Intel NUC as onboard computer. I have a uart cable from pixhawk telem 2 to NUC. Curiously I have not been able to connect to FCU over /dev/ttyUSB0 (even though I can confirm that is my uart cable by unplugging and checking ls /dev/ttyUSB0). When I run:

roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:921600

Everything seems to launch smoothly, but then it hangs after "[ INFO] [1524090355.377067007]: MAVROS started. MY ID 1.240, TARGET ID 1.1".

However, if I use UDP:

roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.50.245:14540" gcs_url:="udp://:14550@192.168.50.159:14550"

I get a hearbeat and then an error message saying "CON: Got HEARTBEAT, connected. FCU: No valid autopilot". It times out before reading any data. I've tried 921600 and 57600 as baud rates for sys_comp in px4 parameters.

Many thanks in advance for your help!

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.7.3

### Node logs

```
uavob@uavob:~/catkin_ws$ roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.50.245:14540" gcs_url:="udp://:14550@192.168.50.159:14550"
... logging to /home/uavob/.ros/log/5065451e-4350-11e8-81b8-c03fd56ce87a/roslaunch-uavob-16260.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uavob:37649/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/conn_heartbeat: 5.0
 * /mavros/conn_timeout: 5.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
[... 367 lines cut ...]
      - 
        key: "Last dt (ms)"
        value: "-0.014620"
      - 
        key: "Mean dt (ms)"
        value: "0.000402"
      - 
        key: "Last system time (s)"
        value: "1524090431.341967821"
      - 
        key: "Time offset (s)"
        value: "9223372036.854705811"
---
```

### Check ID

```
ERROR. I got 1 addresses, but not your target 1:1

---
Received 336 messages, from 1 addresses
sys:comp   list of messages
  1:240   0, 2, 43, 76, 111, 21
```

vooon, 2018-04-19:
Looks like you got loop and receive heartbeats from mavros.

```yml
      - 
        key: "Vehicle type"
        value: "Onboard companion controller"
      - 
        key: "Autopilot type"
        value: "No valid autopilot"
```

sudosurf, 2018-04-19:
Do I need to flash my pixhawk with a different version of the px4 software or is there a problem with my parameters that is resulting in the "No valid autopilot"? Thank you

TSC21, 2018-04-19:
> Looks like you got loop and receive heartbeats from mavros.

You have your answer here. You somehow have a loopback on your system that you need to fix.

sudosurf, 2018-04-19:
If I understand the loopback issue correctly, I am referencing the same port multiple times in mavros and/or my companion computer is referencing the wrong port on the FCU. From reading forums, it looks like I might have better luck trying tdp.  Any other tips on identifying correct ports and protocols? I have had trouble finding documentation on setting FCU parameters for system id, system component, target id, and target component, could this also be an issue? Thank you for your patience and assistance.

sudosurf, 2018-04-30:
For anyone with a similar issue, I made a new USB to UART cable and I was able to get mavros topics after connection. This was a simple cable issue.

closed 2018-04-30 by sudosurf