mavlink/mavros #1017: MAVROS regression connecting to 3DR Solo drones over WIFI

sondreengebraten, 2018-04-23:
### Issue details

Unable to connect to autopilot on 3DR Solo drone using UDP over controller access point link. This used to work perfectly, but upon updating MAVROS to 0.23.3 this breaks. I need to connect to the autopilot to set ardupilot parameters. Currently heartbeats are received from the autopilot, but no information appears to be sent back to the autopilot from the host computer. This causes parameter list to never be loaded, home position is never set.  Note that this is clearly a regression issue or perhaps something to do with a new way of configuring MAVROS, as 15 3DR Solos have already been successfully configured this way using an older MAVROS version. MAVROS was installed using APT. Any suggestions would be most welcome.

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Linux Mint 18.3 / Ubuntu: 16.04

### Autopilot type and version

[ x] ArduPilot

Version: ArdupilotSolo-2.3.0-1

### Node logs

_roslaunch mavros apm.launch fcu_url:=udp://0.0.0.0@10.1.1.10:14560 gcu_url:=udp://@localhost_

```started roslaunch server http://mp005:35245/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://0.0.0.0@10....
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
[... 368 lines cut ...]
        key: "Noice level"
        value: "0"
      - 
        key: "Remote noice level"
        value: "0"
      - 
        key: "Rx errors"
        value: "4"
      - 
        key: "Fixed"
        value: "0"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1273 messages, from 2 addresses
sys:comp   list of messages
 10:0     109
  1:1     0, 1, 2, 129, 136, 137, 152, 150, 24, 27, 29, 30, 33, 35, 36, 165, 42, 178, 62, 193, 74, 163, 253, 116, 252, 125
```

khancyr, 2018-04-23:
hello, change protocol to mavlink v1 and it should works

sondreengebraten, 2018-04-23:
How do I do that?

sondreengebraten, 2018-04-23:
If this is a problem others might run into too, I suggest including the following pull request #1018

sondreengebraten, 2018-04-23:
@khancyr  Worked like a charm :) Thank you!

TSC21, 2018-05-04:
Fixed in #1018.

closed 2018-05-04 by TSC21