mavlink/mavros #1019: no mavros for ros melodic yet?

fnoop, 2018-04-28:
----

### Issue details

Ubuntu 18.04 LTS (Bionic) has now been released.  ROS Melodic looks to be the first distro to support it and compiles/runs fine, but when trying to compile mavros I get this:

```
The following not released packages/stacks will be ignored: mavros
No packages/stacks left after ignoring not released
```

### MAVROS version and platform

Mavros: ?
ROS: Melodic
Ubuntu: 18.04

TSC21, 2018-04-28:
We still have not released MAVROS to Melodic, no. Also we are planning to extend the CI system to multiple releases, so I guess that would be something done together.

fnoop, 2018-04-28:
Is there work to be done to release it, or is it a matter of registering it?  Ubuntu 18.04 is now the current LTS version so would be great to be able to run mavros on it, even if melodic is still in beta.

TSC21, 2018-04-28:
@vooon ^

vooon, 2018-04-29:
I think you should be able to compile master on Melodic, but right now it lack most of dependencies. E.g. `mav_msgs`.

TSC21, 2018-04-29:
@vooon, @fnoop as already tried and it built on master. What about we do a release of mavros, mavros_extras and mavros_msgs to Melodic in the mean time?

vooon, 2018-04-30:
We can't release until **all** dependency packages will be released. That is why i don't like add deps on non core packages. `mav_msgs` several times "forgot" to release.

vooon, 2018-05-07:
Solved by https://github.com/ros/rosdistro/pull/17773 .

closed 2018-05-07 by vooon

fnoop, 2018-05-07:
I still get this:
```
Notice: /Stage[main]/Maverick_ros/Exec[catkin_make_mavros]/returns: mavros: No definition of [geographiclib] for OS version [bionic]
Notice: /Stage[main]/Maverick_ros/Exec[catkin_make_mavros]/returns: mavlink: No definition of [python-future] for OS version [bionic]
```

fnoop, 2018-05-07:
If I remove geographiclib from package.xml and continue, I get this:
```
[ 16%] Building CXX object CMakeFiles/mavconn.dir/src/mavlink_helpers.cpp.o
/srv/maverick/var/build/ros_catkin_ws/src/mavros/libmavconn/src/mavlink_helpers.cpp: In static member function ‘static void mavconn::MAVConnInterface::init_msg_entry()’:
/srv/maverick/var/build/ros_catkin_ws/src/mavros/libmavconn/src/mavlink_helpers.cpp:30:2: error: ‘logDebug’ was not declared in this scope
  logDebug("mavconn: Initialize message_entries map");
  ^~~~~~~~
/srv/maverick/var/build/ros_catkin_ws/src/mavros/libmavconn/src/mavlink_helpers.cpp:30:2: note: suggested alternative: ‘logbl’
  logDebug("mavconn: Initialize message_entries map");
  ^~~~~~~~
  logbl
CMakeFiles/mavconn.dir/build.make:62: recipe for target 'CMakeFiles/mavconn.dir/src/mavlink_helpers.cpp.o' failed
make[2]: *** [CMakeFiles/mavconn.dir/src/mavlink_helpers.cpp.o] Error 1
CMakeFiles/Makefile2:323: recipe for target 'CMakeFiles/mavconn.dir/all' failed
make[1]: *** [CMakeFiles/mavconn.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

vooon, 2018-05-07:
That looks like you have problem with `libconsole-bridge-dev`.

fnoop, 2018-05-08:
Related: https://github.com/mavlink/mavros/issues/985
I'm already using catkin_make_isolated

fnoop, 2018-05-08:
https://github.com/ros/ros-overlay/issues/509
If I prefix logDebug, LogError etc with CONSOLE_BRIDGE_, then mavros compiles