mavlink/mavros #1024: Can a 2D SLAM pose be integrated into /mavros/local_position/pose ?

tiralonghipol, 2018-05-09:
### Issue details

Hi. I'm working on autonomous navigation (no gps) and I was wondering:

- in offboard mode, to get a perfect drone Pose, how can I integrate the slam_out_pose (of hector_slam, for example) with the /mavros/local_position/pose (which, if I understand well, it's only "imu-based")?
Is this feature supported? 

- when I send commands to /mavros/setpoint_velocity/cmd_vel which is the reference frame?

I have the 2 poses (slam and mavros), as you can see in the screen below

[image]

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: v1.7.3

####################################################################
### Node logs
####################################################################

auto-starting new master
process[master]: started with pid [17374]
ROS_MASTER_URI=http://172.16.3.8:11311

setting /run_id to d2be007e-535d-11e8-97bb-d4bed98eab61
process[rosout-1]: started with pid [17388]
started core service [/rosout]
process[mavros-2]: started with pid [17406]
[ INFO] [1525852307.490543579]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1525852307.492333693]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1525852307.493050174]: GCS bridge disabled
[ INFO] [1525852307.507584718]: Plugin 3dr_radio loaded
[ INFO] [1525852307.510055892]: Plugin 3dr_radio initialized
[ INFO] [1525852307.510181427]: Plugin actuator_control loaded
[ INFO] [1525852307.513733890]: Plugin actuator_control initialized
[ INFO] [1525852307.516313599]: Plugin adsb loaded
[ INFO] [1525852307.519913753]: Plugin adsb initialized
[ INFO] [1525852307.520105616]: Plugin altitude loaded
[ INFO] [1525852307.521501976]: Plugin altitude initialized
[ INFO] [1525852307.521613563]: Plugin cam_imu_sync loaded
[ INFO] [1525852307.522684642]: Plugin cam_imu_sync initialized
[ INFO] [1525852307.522868879]: Plugin command loaded
[ INFO] [1525852307.528125614]: Plugin command initialized
[ INFO] [1525852307.528248648]: Plugin debug_value loaded
[ INFO] [1525852307.533449692]: Plugin debug_value initialized
[ INFO] [1525852307.533507572]: Plugin distance_sensor blacklisted
[ INFO] [1525852307.533725271]: Plugin fake_gps loaded
[ INFO] [1525852307.553231620]: Plugin fake_gps initialized
[ INFO] [1525852307.553483425]: Plugin ftp loaded
[ INFO] [1525852307.563319680]: Plugin ftp initialized
[ INFO] [1525852307.563505917]: Plugin global_position loaded
[ INFO] [1525852307.583177354]: Plugin global_position initialized
[ INFO] [1525852307.583373280]: Plugin hil loaded
[ INFO] [1525852307.599667852]: Plugin hil initialized
[ INFO] [1525852307.599896514]: Plugin home_position loaded
[ INFO] [1525852307.604369774]: Plugin home_position initialized
[ INFO] [1525852307.604623532]: Plugin imu loaded
[ INFO] [1525852307.612871516]: Plugin imu initialized
[ INFO] [1525852307.613076858]: Plugin local_position loaded
[ INFO] [1525852307.618958154]: Plugin local_position initialized
[... 205 lines cut ...]
        value: "-1.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "9019"
      - 
        key: "Frequency (Hz)"
        value: "10.000476"
      - 
        key: "Last dt (ms)"
        value: "2.635106"
      - 
        key: "Mean dt (ms)"
        value: "0.004569"
      - 
        key: "Last system time (s)"
        value: "1495.383272000"
      - 
        key: "Time offset (s)"
        value: "1525851714.332882404"

TSC21, 2018-05-10:
> - in offboard mode, to get a perfect drone Pose, how can I integrate the slam_out_pose (of hector_slam, for example) with the /mavros/local_position/pose (which, if I understand well, it's only "imu-based")?
Is this feature supported? 

Offboard mode is a control mode. What you probably want is fuse your SLAM output on the onboard estimator of the flight controller. For that, you need to use the `vision_pose_estimate` of `mavros_extras` and have PX4 ready to fuse that data (rather than setting the proper parameters in EKF2 or LPE depending of which estimator you are using). `/mavros/local_position/pose` is not IMU based, it is the estimation of the vehicle position parsed from `LOCAL_POSITION_NED` Mavlink msg.

> - when I send commands to /mavros/setpoint_velocity/cmd_vel which is the reference frame?

That depends if you are sending commands on the local (`world` or `odom` on ROS) or body frames (`base_link` on ROS). For the first, it's `LOCAL_NED`, for the second `BODY_NED`.

tiralonghipol, 2018-05-10:
Thank you so much for the reply! Still a little bit confused on many aspects:
- Assuming that I would like to use the "vision_pose_estimate", is it enough to publish on /mavros/local_position/odom? If not, what is the topic name? 

- From the screen I posted above, you can see that the pose coming from LOCAL_POSITION_NED (pink) and slam_out_pose (green) are very different. Can you guess where is the problem? Maybe Reference frames ?

Sorry for all these questions but I've struggling months on this topic and I think it's time to make all the necessary clarifications.

Thanks again!

TSC21, 2018-05-10:
> - Assuming that I would like to use the "vision_pose_estimate", is it enough to publish on /mavros/local_position/odom? If not, what is the topic name?

`/mavros/local_position/odom` is a publisher, not a subscriber. So you don't publish to it, you listen to the topic. You have to publish to `/mavros/vision_pose/pose` given that `slam_out_pose` publishes `geometry_msgs/PoseStamped` msgs.

> - From the screen I posted above, you can see that the pose coming from LOCAL_POSITION_NED (pink) and slam_out_pose (green) are very different. Can you guess where is the problem? Maybe Reference frames ?

If you are not publishing any SLAM data to the flight controller, you can't expect that the data matches.

tiralonghipol, 2018-05-10:
ok so I guess the plan is 
- insert inside px4_pluginlist.yaml "vision_pose_estimate" (and "mocap_pose_estimate" ?)
- publish the slam_out_pose on /mavros/vision_pose/pose

do I have to change anything inside px4_config.yaml?

TSC21, 2018-05-10:
> - insert inside px4_pluginlist.yaml "vision_pose_estimate" (and "mocap_pose_estimate" ?)

No, it is already up by default.

> do I have to change anything inside px4_config.yaml?

Unless you want to listen to transforms, no.

TSC21, 2018-05-10:
> If I can't see any changes could be related to the update frequency? I mean that the slam pose (since I'm using a cheap lidar) could maybe be too slow if compared to the pixhawk..

Not really. Have you set PX4 param [`EKF2_AID_MASK`](https://dev.px4.io/en/advanced/parameter_reference.html#EKF2_AID_MASK) to something like 22 (bits 3 & 4 set)? That's assuming that you are using PX4 Firmware on default settings. To use LPE+ATQ, you would have to set [`SYS_MC_EST_GROUP`](https://dev.px4.io/en/advanced/parameter_reference.html#SYS_MC_EST_GROUP) to 1 and then change the respective parameter to LPE which is [`LPE_FUSION`](https://dev.px4.io/en/advanced/parameter_reference.html#LPE_FUSION).

tiralonghipol, 2018-05-10:
> Have you set PX4 param EKF2_AID_MASK to something like 22 (bits 3 & 4 set)?

Yes, I have set this parameter in QGroundControl, it corresponds to 24

> set SYS_MC_EST_GROUP to 1 and then change the respective parameter of LPE which is LPE_FUSION. 

When I try to change this parameter from ekf2 to lpe, after every reboot the value keeps coming back to ekf, so I cannot set it to lpe. Quite strange no?

I'm uxing PX4 stable version 1.7.3

TSC21, 2018-05-10:
> When I try to change this parameter from ekf2 to lpe, after every reboot the value keeps coming back to ekf, so I cannot set it to lpe. Quite strange no?

Doesn't matter to the issue. If you are setting `EKF2_AID_MASK` to 24 that means you are fusing flow also. You have a optical flow module attached, right?

TSC21, 2018-05-10:
Also I advise to step up to the latest master, cause EKF2 has been suffering some changes that can improve the estimation performance.

tiralonghipol, 2018-05-11:
ok so now I have even more questions:

- I have git cloned mavros inside my catkin_ws. In order to update it to the latest version, do I have to delete the entire folder or is there another way?

- How is the pix4flow module related to the output slam pose coming from a 2D laser scanner? I'm getting a little bit confused, hope you can clarify me this point

> You have a optical flow module attached, right?

Sure, it is attached to the i2c port, as the guide suggests, but I never realized well how to use it (I updated the pix4flow firmware stable version on QGC too, before you ask)

Really sorry to bother you so much, little newbe here :)

YCH188, 2019-03-08:
@TSC21 Hello, I have a question, if I publish the x,y,z from slam to topic /mavros/vision_pose/pose, and publish the position which I want the UAV to go through mavros/setpoint_position/local topic, will the UAV fly according to what I publish in /mavros/vision_pose/pose, or it will still fly according to /mavros/local_position/pose? Thanks a lot !