mavlink/mavros #1025: Mode change using MAVROS

TrishantRoy, 2018-05-09:
### Issue details
I am using rc/override to control the quad. I am unable to change mode of the quad (stabilise to loiter) through code. I have tried using setMode rosservice. Can setMode be used for APM flight stack?
```
    ros::ServiceClient cl = n.serviceClient<mavros_msgs::SetMode>("/mavros/set_mode");
    mavros_msgs::SetMode srv_setMode;
    srv_setMode.request.base_mode = 0;
    srv_setMode.request.custom_mode = "ALT_HOLD";
    if(cl.call(srv_setMode)){
        ROS_INFO("setmode send ok %d value:", srv_setMode.response.mode_sent);
    }else{
        ROS_INFO("Failed SetMode");
    }
```

### MAVROS version and platform

Mavros: 0.24.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version
APM 3.5.5

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch

... logging to /home/iarc/.ros/log/0cebc4da-5399-11e8-b159-a434d90f8f75/roslaunch-iarc-nuc-2124.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://iarc-nuc.local:39917/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
[... 311 lines cut ...]
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
---

```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 139 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 74, 77, 178, 182, 253, 30

```

TrishantRoy, 2018-05-09:
Is there a way to check the current mode?

Thanks in advance.

lamping7, 2018-05-11:
`mavros/state` -- http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.sys_status

closed 2018-05-23 by TrishantRoy