mavlink/mavros #1027: Desidered feature: MultiUAV scenario with messages published on different topics

quattrinili, 2018-05-16:
### Issue details

The setup is as follows:
- two pixhawk PX4 running ArduRover with an RFD900+ each. Each of them has a different sysid.
- one laptop as base station with an RFD900+, running MAVROS
The radios are on the same channel and the base station can see the two pixhawk mavlink messages.

The issue is the fact that the MAVROS node does not differentiate between the two pixhawk, as such, on the same topic, messages from the two pixhawk are published. As such, the desired feature is to distinguish between the two.

Before looking at the code, I tried to use tgt_sys_id, however, setting it to the sys_id of one of them does not filter the messages coming from the other one.

Looking at the code there could be different solutions, including:
a) filtering within `plugin_route_cb`. This requires running multiple MAVROS node, which might not be ideal in terms of resource usage.
b) including a prefix in the string for the frame in the header file, which however if some messages do not have a header file, it might not be possible.
c) creating multiple publishers, according to the number of different sys_id detected.

I will probably test a) as it seems to be relatively easy, however c) probably is the best solution among the ones I listed.

### MAVROS version and platform

Mavros: 0.24.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: 3.2.1

quattrinili, 2018-05-16:
I forked, implemented solution a), and tested it, for the tag 0.24.0 given that it works out of the box with ROS kinetic. The solution can be found here:

https://github.com/quattrinili/mavros/tree/0.24.0-tgtsysidfilter

AlexisTM, 2018-05-23:
I am pretty sure (c) is not viable for this project as Mavros is the HAL for one copter; My solution is having Mavros running on each copter instead of the master;

Doolly, 2018-06-17:
is this branch has solved https://github.com/mavlink/mavros/issues/872 this issue?

ArrEssJay, 2018-08-22:
I have a similar problem. mavros is connected to the FCU and a GCS via mavlink-router. Different UAVs are using mav system ID values. mavros is not filtering out messages for other system IDs. I haven't had chance to test the solution above filtering based on the target ID but in principle this should solve the issue.