mavlink/mavros #1028: WP: upload failed: generic error / not accepting mission commands at all right now

Ahrovan, 2018-05-20:
### Issue details
this error occurs:
_WP: upload failed: generic error / not accepting mission commands at all right now_
and mavros say -> **success: False** for "call /mavros/mission/push"

### MAVROS version and platform
Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04
PX4

### Node logs
```
user@user:~/src/Firmwarerosservice call /mavros/mission/push "start_index: 0
waypoints:
- {frame: 3, command: 22, is_current: false, autocontinue: false, param1: 0.0, param2: 0.0,
  param3: 0.0, param4: 0.0, x_lat: 47.397734, y_long: 8.5455834, z_alt: 10.0}" 
success: True
wp_transfered: 1
user@user:~/src/Firmware$ rosservice call /mavros/mission/push "start_index: 0
waypoints:
- {frame: 3, command: 22, is_current: false, autocontinue: false, param1: 0.0, param2: 0.0,
  param3: 0.0, param4: 0.0, x_lat: 47.397734, y_long: 8.5455834, z_alt: 10.0}" 
success: False
wp_transfered: 1
user@user:~/src/Firmware$ 
```
I try this steps with ros node and push with pushWaypointList() function (same problem occur)

### Diagnostics

```
user@user:~/src/Firmware$ rostopic echo -n1 /diagnostics

header: 
  seq: 204
  stamp: 
    secs: 200
    nsecs: 500000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Received packets:"
        value: "10724"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "230"
      - 
        key: "Tx sequence number:"
        value: "247"
      - 
        key: "Rx total bytes:"
        value: "3296163"
      - 
        key: "Tx total bytes:"
[... 407 lines cut ...]
[ INFO] [1526806058.186130967, 1.298000000]: udp0: Remote address: 127.0.0.1:14557
ERROR [mavlink] [timesync] Hard setting offset.
[ INFO] [1526806058.306390242, 1.418000000]: IMU: High resolution IMU detected!
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[ INFO] [1526806059.092767985, 2.202000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1526806059.094233565, 2.202000000]: IMU: High resolution IMU detected!
[ INFO] [1526806059.214715775, 2.322000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1526806060.092548446, 3.204000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1526806060.092601867, 3.204000000]: VER: 1.1: Flight software:     01070300 (00000000D832D2E0)
[ INFO] [1526806060.092650470, 3.204000000]: VER: 1.1: Middleware software: 01070300 (00000000D832D2E0)
[ INFO] [1526806060.092732990, 3.204000000]: VER: 1.1: OS software:         040d00ff (0000000000000000)
[ INFO] [1526806060.092762786, 3.204000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1526806060.092775607, 3.204000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1526806060.092802998, 3.204000000]: VER: 1.1: UID:                 0000000100000002
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
[ INFO] [1526806069.105368787, 12.202000000]: HP: requesting home position
[ INFO] [1526806074.109650133, 17.202000000]: WP: mission received
[ WARN] [1526806084.618793621, 27.702000000]: TM: Clock skew detected (-0.090642969 s). Hard syncing clocks.
ERROR [mavlink] [timesync] Hard setting offset.
[ INFO] [1526806145.588104445, 88.610000000]: WP: mission sended
WARN  [mavlink] WPM: Unexpected mission type (48 0)
[ERROR] [1526806148.917659488, 91.938000000]: WP: upload failed: generic error / not accepting mission commands at all right now
WARN  [mavlink] Operation timeout
```

Ahrovan, 2018-05-20:
related Mission upload broken on mavros 0.23.0 with PX4 #957

TSC21, 2018-05-20:
There's a fix already pushed upstream for that issue.

LorenzMeier, 2018-05-20:
Where is that fix? In MAVROS? Do we need a release? Do you have a link?

Ahrovan, 2018-05-20:
Announced here https://github.com/mavlink/mavros/commit/415b789e13524e7c91b6eaad2373ac456340385d#diff-88072a420ceca1c0537dd02f9e7bc718
I'm testing with PX4 and announcing the result a few minutes later

TSC21, 2018-05-20:
#957 was closed with https://github.com/mavlink/mavros/commit/415b789e13524e7c91b6eaad2373ac456340385d. If @Ahrovan considers it related, then he should try the upstream version and check if it is fixed.

Ahrovan, 2018-05-20:
@TSC21 problem solved with upstream version of mavros and PX4

closed 2018-05-20 by TSC21