mavlink/mavros #1029: How to fly a swarm of drones in one coordinate system? (on offboard mode)

GigaFlopsis, 2018-05-22:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform
Mavros: 0.24.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version
[  ] ArduPilot
[ x ] PX4
Version: 1.7.3
pixhawk: 2.6.8

Hello! 
I want to control a swarm of drones via offboard mode. Is there a have method for common one origin for a group of drones?
Ideally, they should have a common local coordinate system.

AlexisTM, 2018-05-23:
Solution 1: use Global setpoints.
Solution 2: use local setpoints and either put tags/camera on the floor to locate the copters on a common frame or compute the transformation between copters using GPS position.

I am using a common reference (tags) that are defined in position.

GigaFlopsis, 2018-05-23:
I'm already working on my first solution. Transform from WGS 84 to NED and back is not a problem, however, I'm not sure that the global position is published too often, even if I increase the frequency in mavros. 

I correctly understood that the firmware does not have the function for set origin relative to the global coordinate system?
I mean, at system startup, the drone considers a local position at zero, can this position be offset ?

AlexisTM, 2018-05-23:
I wanted to say to change the home position; but for control reasons, I don't think it moves the local_setpoint origin.

You can get the frame by computing the current HOME POSITION as (0,0,0) to the other copters; This is a static frame! (don't have to recompute it)

GigaFlopsis, 2018-05-28:
Hello again. I decided to use global_setpoint. I noticed that the global coordinates are strongly floating , on board is also used px4flow. Does the global position take into account this sensor when it publishes the coordinates?

TSC21, 2018-05-31:
@GigaFlopsis this may be of your interest: https://github.com/PX4/Firmware/issues/7368