mavlink/mavros #1032: vision pose problem

gajena, 2018-05-23:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am using aruco markers for position feedback and publishing to topic `mavros/vision_pose/pose` and I have also checked the `mavros/local_position/pose` and data is same as `vision_pose`.

---
*  Read the complete development guide for offboard control.
---
* Now I am setting the `local_position` setpoint for `x and y` is zero

---
* But there is nothing happening like it was doing nothing to maintain its position.

---
* Altitude controller is working fine, mainly facing the problem with roll and pitch

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4 1.7.3

Version: ?3.7.1?

TSC21, 2018-05-31:
@gajena without:
1. A code snippet showing how are you publishing the data to to `mavros/vision_pose/pose`;
2. A code snippet showing how are you publishing the setpoints to `mavros/setpoint_position/local`;
3. Most important: logs!

will be difficult to debug your problem.

TSC21, 2018-07-13:
Closing as stale.

closed 2018-07-13 by TSC21

YCH188, 2019-03-08:
@gajena Have you solve the problem? Should I publish the point to /mavros/setpoint_position/local to control the flight.

gajena, 2019-03-10:
@YCH188 No, I have not figured it out. Instead, I implemented a high-level controller on the onboard computer from [mav_control_rw](https://github.com/ethz-asl/mav_control_rw).
If, you really want to implement on PX4. I think this can help  [PX4_avoidance](https://github.com/PX4/avoidance).