mavlink/mavros #1033: setpoint_attitude not working with new mavros update (probably)

siddhu45, 2018-05-24:
----

### Issue details

_The setpoint_attitude topic is not publishing to the `mavros/setpoint_attitude/target_attitude` topic after the latest mavros update. I know this because the px4_config file in the mavros launch folder has changed from the previous version. When I start the gazebo simulation it gives an error message:_ ODOM: invalid body frame orientation "frd".

Although, the position & velocity setpoint publishers for px4 still works, but the attitude setpoint isn't. I tried running my previous version of mavros and it works properly, I checked from `rqt_graph`. So, if it is mavros side issue, what can be done to resolve this?

Thanks

Mrrrbel, 2018-05-31:
I had similar issue (ODOM: invalid body frame orientation "frd"). I downgraded to 0.24.0, and it went away.

TSC21, 2018-05-31:
For `ODOM: invalid body frame orientation "frd".` - fix here: https://github.com/mavlink/mavros/pull/1041.

> The setpoint_attitude topic is not publishing to the mavros/setpoint_attitude/target_attitude topic after the latest mavros update. I know this because the px4_config file in the mavros launch folder has changed from the previous version.

This is far from being helpful. You need to describe what are you trying to do and what versions are you comparing (meaning what release are you using and what release were you using). Having `px4_config` changed does not mean a problem unless you actually follow the changes required. If you did see the change on the launch folder, you should have checked also for the topic naming, which have changed. I recall that for publishing to `mavros/setpoint_attitude/attitude` (it changed from `mavros/setpoint_attitude/target_attitude` to `mavros/setpoint_attitude/target_attitude`) you also need to publish to `mavros/setpoint_attitude/thrust`.

TSC21, 2018-05-31:
@Mrrrbel regarding what you report: fix in #1041. Should get in soon and then a new release will come out.

closed 2018-05-31 by TSC21

siddhu45, 2018-06-02:
@TSC21 Yea I was not clear on the versions, apologies! I should have described in more detail but since that day I was out of lab for a few days and I later remembered the issue I faced, so was not able to confirm the versions. 

>  I recall that for publishing to mavros/setpoint_attitude/attitude (it changed from `mavros/setpoint_attitude/target_attitude` to `mavros/setpoint_attitude/target_attitude`) you also need to publish to `mavros/setpoint_attitude/thrust`.

Yes, I am aware of this change till the last version of mavros which I used and was publishing to both the topics for the attitude setpoint to work. 

Thanks for the fix though!

TSC21, 2018-06-02:
Sorry, I meant changed from `mavros/setpoint_attitude/target_attitude` to `mavros/setpoint_attitude/attitude`. Is it working for you now?

TSC21, 2018-06-02:
@vooon the bug is not related to this issue, but rather to a bug introduced by d738777.

siddhu45, 2018-06-02:
Oh, so it changed back to `mavros/setpoint_attitude/attitude`? From version 0.25 is it? Because the version I was using prior to that (not sure which one - need to check) was still using `mavros/setpoint_attitude/target_attitude` (I think it changed to this topic around 0.19/0.20). And I haven't checked the latest fix, will be back in the lab after Monday, so will check and notify!

TSC21, 2018-06-02:
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L82