mavlink/mavros #1035: No response to ~set_stream_rate service call

szebedy, 2018-05-24:
### Issue details

I am trying to increase the update rate of the IMU using the ~set_stream_rate service call. I tried increasing and decreasing the rate as well, unfortunately without success.

I have tried the following:
In a terminal: `rosservice call /mavros/set_stream_rate 1 200 1` (also tried it with 0 200 1, 0 10 1, 0 10 0 etc.)
In C++:
```cpp
#include <mavros_msgs/StreamRate.h>

    ros::ServiceClient stream_rate_client;
    ros::NodeHandle nh;
    stream_rate_client = nh.serviceClient<mavros_msgs::StreamRate>("/mavros/set_stream_rate");

    mavros_msgs::StreamRate streamRate;
    streamRate.request.stream_id = 1;
    streamRate.request.message_rate = 200;
    streamRate.request.on_off = true;
    stream_rate_client.call(streamRate);
```
None of these worked, have no idea why. I also don't know how to proceed with debugging.

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.7.3

### Node logs

```
$ rosrun mavros mavros_node _fcu_url:=tcp://127.0.0.1:5760 _system_id:=2
[ INFO] [1527166276.947757128]: FCU URL: tcp://127.0.0.1:5760
[ INFO] [1527166276.950742270]: tcp0: Server address: 127.0.0.1:5760
[ INFO] [1527166276.951111877]: GCS URL: udp://@
[ INFO] [1527166276.951511896]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1527166276.987531613]: Plugin 3dr_radio loaded
[ INFO] [1527166276.992448915]: Plugin 3dr_radio initialized
[ INFO] [1527166276.992688770]: Plugin actuator_control loaded
[ INFO] [1527166277.007269984]: Plugin actuator_control initialized
[ INFO] [1527166277.017943503]: Plugin adsb loaded
[ INFO] [1527166277.034735271]: Plugin adsb initialized
[ INFO] [1527166277.034972938]: Plugin altitude loaded
[ INFO] [1527166277.039048387]: Plugin altitude initialized
[ INFO] [1527166277.039256704]: Plugin cam_imu_sync loaded
[ INFO] [1527166277.041790812]: Plugin cam_imu_sync initialized
[ INFO] [1527166277.042051917]: Plugin command loaded
[ INFO] [1527166277.065202700]: Plugin command initialized
[ INFO] [1527166277.065648021]: Plugin debug_value loaded
[ INFO] [1527166277.087290239]: Plugin debug_value initialized
[ INFO] [1527166277.087577270]: Plugin distance_sensor loaded
[ WARN] [1527166277.091093971]: DS: plugin not configured!
[ INFO] [1527166277.091226799]: Plugin distance_sensor initialized
[ INFO] [1527166277.091482266]: Plugin fake_gps loaded
[ INFO] [1527166277.138180477]: Plugin fake_gps initialized
[ INFO] [1527166277.138560459]: Plugin ftp loaded
[ INFO] [1527166277.171255804]: Plugin ftp initialized
[ INFO] [1527166277.171585385]: Plugin global_position loaded
[ INFO] [1527166277.235530135]: Plugin global_position initialized
[ INFO] [1527166277.235808240]: Plugin hil loaded
[ INFO] [1527166277.301855916]: Plugin hil initialized
[ INFO] [1527166277.302150084]: Plugin home_position loaded
[ INFO] [1527166277.318906651]: Plugin home_position initialized
[... 310 lines cut ...]
        value: "17.91"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "100.0"
---

```

### Check ID

```
WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
OK. I got messages from 1:1.

---
Received 9307 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 266, 140, 141, 147, 24, 30, 31, 32, 33, 36, 74, 76, 83, 85, 87, 230, 231, 105, 111, 241, 242, 245

```

dagar, 2018-05-25:
PX4 has support for MAV_CMD_GET_MESSAGE_INTERVAL and  MAV_CMD_SET_MESSAGE_INTERVAL.

https://mavlink.io/en/messages/common.html#MAV_CMD_GET_MESSAGE_INTERVAL

vooon, 2018-05-25:
Yes, but that iface is not yet supported by mavros (requires new rosapi call).

szebedy, 2018-06-01:
@vooon , @dagar Thanks for pointing me to the right direction. Could you suggest a way to get the MAV_CMD_SET_MESSAGE_INTERVAL mavlink command to PX4?

I tried `rosrun mavros mavcmd int 511 105 10000 100 1 1 1 1`, but no success. The numbers stand for the following:
511: https://mavlink.io/en/messages/common.html#MAV_CMD_GET_MESSAGE_INTERVAL
105: https://github.com/PX4/Firmware/blob/4453e4201b7a245cff52beeb38a293161aea4c48/Tools/mavlink_px4.py#L506
1 1 1 1: Had to fill up with something, otherwise got `mavcmd int: error: too few arguments`

szebedy, 2018-06-01:
Also increased the IMU rate with the mavlink module of PX4 `mavlink stream -u 14556 -s HIGHRES_IMU -r 100`. Still the rate stays 50 when I check it with `rostopic hz /mavros/imu/data`. What needs to be done outside of mavros so that is publishes the `/mavros/imu/data` topic faster?

vooon, 2018-06-01:
Try also command long, `mavcmd long 511 105 1000 0 0 0 0 0 0`

szebedy, 2018-06-01:
Thanks for the quick reply @vooon 

Tried `rosrun mavros mavcmd long 511 105 1000 0 0 0 0 0`, unfortunately the rate is still 50.

What triggers mavros to publish on `/mavros/imu/data`?

TSC21, 2018-06-01:
@szebedy the Aero is streaming the data through UDP right? What's the port it is using? Why are you plugging through TCP instead? `mavlink stream -u 14556 -s HIGHRES_IMU -r 100` will only work if the Aero is actually streaming through that UDP port.

> What triggers mavros to publish on `/mavros/imu/data`? I've found that the data is published [here](https://github.com/mavlink/mavros/blob/76b4b677864d0d6a3d20a4c719a2eb37b9dcabd6/mavros/src/plugins/imu.cpp#L346) in the function `handle_attitude_quaternion`, but I haven't found what triggers that function.

It triggers when MAVROS receives a `HIGHRES_IMU` msg from the stream.

szebedy, 2018-06-01:
@TSC21 Very good remark, the Aero Flight Controller is actually streaming the data through a serial port (/dev/ttyS1) to the compute board. However, to make things "easier", the compute board has a proxy which exposes the serial port as a network socket (port 5760).

Therefore I tried `mavlink stream -d /dev/ttyS1 -s HIGHRES_IMU -r 100`, but unfortunately the streaming rate remains 50 :(

TSC21, 2018-06-01:
> the compute board has a proxy which exposes the serial port as a network socket (port 5760).

Then the problem may be on the proxy. I would guarantee that the bottleneck is not there.

szebedy, 2018-06-01:
Thank you so much @vooon, that did the trick!!! So the current solutions are:

1. Execute the following in QGC Mavlink Console:
```
mavlink stream -d /dev/ttyS1 -s HIGHRES_IMU -r 100
mavlink stream -d /dev/ttyS1 -s ATTITUDE_QUATERNION -r 100
```

2. In a terminal
```
rosrun mavros mavcmd long 511 105 10000 0 0 0 0 0
rosrun mavros mavcmd long 511 31 10000 0 0 0 0 0
```
Is it possible to include these two commands in a ROS launch file (or in a C++ ROS node)?

TSC21, 2018-06-01:
> Is it possible to include these two commands in a ROS launch file (or in a C++ ROS node)?

Just put:
```
mavlink stream -d /dev/ttyS1 -s HIGHRES_IMU -r 100
mavlink stream -d /dev/ttyS1 -s ATTITUDE_QUATERNION -r 100
```
in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4070_aerofc, reprogram your Aero board and reboot it. And you won't need to even use the ROS commands.
Closing this then.

closed 2018-06-01 by TSC21

TSC21, 2018-06-01:
Also just a remark:
If you want to get data from `mavros/imu/data_raw`, you just need to increase the stream of `HIGHRES_IMU`. If you want to get from `mavros/imu/data` (filtered attitude), you just increase from `ATTITUDE_QUATERNION`. You were simply getting the data from the wrong topic in the first place. if you issued `rostopic hz /mavros/imu/data_raw`, you would have saw the increase on the data rate.

szebedy, 2018-06-02:
> Just put:
> 
> mavlink stream -d /dev/ttyS1 -s HIGHRES_IMU -r 100
> 
> or
> 
> mavlink stream -d /dev/ttyS1 -s ATTITUDE_QUATERNION -r 100
> 
> in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4070_aerofc, reprogram your Aero board and reboot it. And you won't need to even use the ROS commands.
> Closing this then.

@TSC21 For some reason this didn't have any effect, any idea why? I inserted `mavlink stream -d /dev/ttyS1 -s ATTITUDE_QUATERNION -r 100` to the end of https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4070_aerofc, built with `make aerofc-v1_default`, then uploaded with `make aerofc-v1_default upload`. The upload completes, I reboot but the rate of `mavros/imu/data` stays 50.

vooon, 2018-06-02:
@TSC21 aero does not have `extras.txt` script feature? Pixhawk can execute extras.txt script from SD card, that allows to setup streams.

TSC21, 2018-06-02:
It doesn't. https://github.com/intel-aero/meta-intel-aero/issues/286

szebedy, 2018-06-02:
> @TSC21 For some reason this didn't have any effect, any idea why? I inserted mavlink stream -d /dev/ttyS1 -s ATTITUDE_QUATERNION -r 100 to the end of [...]

Maybe it's overwritten by this line: https://github.com/PX4/Firmware/blob/ab279d5fe9e014cec26add8a56bd5e676152b1c5/src/modules/mavlink/mavlink_main.cpp#L2042

My current working solution is to replace `configure_stream("ATTITUDE_QUATERNION", 50.0f);` to `configure_stream("ATTITUDE_QUATERNION", 100.0f);`