mavlink/mavros #1036: mavcmd sethome command does not work?

Halmoni100, 2018-05-25:
### Issue details

I'm currently running Gazebo with the PX4 SITL interacting with mavros.  When running "rosrun mavros mavcmd -n [mavros node] sethome ...", the command returns with no output, and it does not like it has any effect (the global_position/local coordinates do not change).  Right now, I'm hacking around this by issuing the ros msg the global_position node expects.

### MAVROS version and platform

Mavros: 0.24.0
ROS: Kinetic
Ubuntu: 16.04

TSC21, 2018-05-31:
`rosrun mavros mavcmd sethome <latitude> <longitude> <altitude>` doesn't work? `sethome` requires coordinates in the global frame.

Halmoni100, 2018-05-31:
The command doesn't have any effect on the global_position/local topic output, which I think it should.

TSC21, 2018-05-31:
No it doesn't. It just sets the position you want as being the home position of the vehicle. So for example, when RTL is issued, the vehicle should fly to that point.
I am closing this since I don't since any problems with this. You can check the current vehicle position home position by checking the topic published by the home_position plugin in MAVROS (since you are using it).

closed 2018-05-31 by TSC21

varunvp, 2019-07-03:
What should I use to see the effect that the original question asks?

shlee853, 2021-09-13:
If I fly indoor without, how can I set my home position? Actually, I don't need my home position, because I'm in local position.

RamanaBotta, 2021-11-29:
rosrun mavros mavcmd sethome 13.589722 80.003377 8.0
Request failed. Check mavros logs. ACK: 1
I dont understand. why it is failing