mavlink/mavros #1037: TCP disconnection exception crashes node

Kiwa21, 2018-05-28:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

tcp2: receive: Connection reset by peer
or 
tcp0: receive: End of file

exception are crashing the node rather than putting it in a "Waiting for connection" state.

(Not sure if it's feasible though...)

### MAVROS version and platform

Mavros: 0.23.3
ROS: Kinetic
Ubuntu: 16.04

### Node logs

```
[ERROR] [1527501069.238680415]: tcp2: receive: Connection reset by peer
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  shutdown: Transport endpoint is not connected
================================================================================REQUIRED process [test/test-1] has died!
process has died [pid 20458, exit code -6, cmd /home/gg/Documents/ws/devel/lib/mavros/mavros_node __name:=test __log:=/home/gg/.ros/log/def9e29a-6258-11e8-93b0-7c5cf8252583/iGCS-iGCS-1.log].
log file: /home/gg/.ros/log/def9e29a-6258-11e8-93b0-7c5cf8252583/test-test-1*.log
Initiating shutdown!
================================================================================
[test/test-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

vooon, 2018-05-28:
Use UDP.

Kiwa21, 2018-05-28:
It might be a silly question, but I lack knowledge here.
Is there a correct way to use UDP with the behavior of tcp-l TCP server?
Define the host / server address rather than the remote / client address?
So "unknown" clients can connect to the node without having their IP address in the first place?

TSC21, 2018-05-29:
@vooon why is this not a bug?

Kiwa21, 2018-05-31:
url as udp://LOCAL_IP:PORT@ seems to work for unknown remote port.
Had trouble before because I was specifying bind_port@remote_port and it wasn't starting.
However, TCP could connect correctly to multiple clients (2) while UDP struggles : 

`[ INFO] [1527778274.568599153]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778275.351006406]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778275.577899158]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778276.353156440]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778276.582778572]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778277.357815495]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778277.592076539]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778278.364133199]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778278.598229753]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778279.366428604]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778279.601171697]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778280.376335598]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778280.518940856]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778281.385266679]: udp0: Remote address: 192.168.1.46:43116
[ INFO] [1527778281.610135537]: udp0: Remote address: 192.168.1.46:34928
[ INFO] [1527778282.385942411]: udp0: Remote address: 192.168.1.46:43116
`