mavlink/mavros #1038: error with mavcmd takeoff and takeoffcur

xup682u03, 2018-05-29:
I was using mavros on rpi3 and testing mavros .

connect to px4

> source ~/catkin_ws/devel/setup.bash

> roscore

> rosrun mavros mavros_node _fcu_url:=/dev/ttyAMA0:921600

----

### MAVROS version and platform

Mavros: 0.18.5
ROS: Kinetic
Ubuntu: 16.04 mate

### Autopilot type and version

[  ] ArduPilot
[O] PX4

### Issue details

I can run `rosrun mavros mavsafety arm` with motor rotating and `rosrun mavros mavcmd set home 0 0 0`  without error.
But i encountered some errors when running  **takeoff** and **takeoffcur** in mavcmd.

PS. I test in room in stead of outside.

error when running `rosrun mavros mavcmd takeoffcur 0 0 50`:
Request failed. Check mavros logs. ACK: 4 
error when running `rosrun mavros mavcmd takeoff 0 0 0 0 50`:
NavSatFix topic not exist 
[image]

the mavros logs:
[image]

The above line with error appear when running `rosrun mavros mavsafety arm` **successfully**.

TSC21, 2018-05-31:
> PS. I test in room in stead of outside.

You cannot takeoff with a position lock, meaning you need to have GPS signal available.

xup682u03, 2018-06-01:
@TSC21 
Thanks for replying.

 This time I test on the roof  and face the same situation:
`rosrun mavros mavsafety arm` with motor rotating 
`rosrun mavros mavcmd set home 0 0 0` without error.
error when running `rosrun mavros mavcmd takeoffcur 0 0 50`:
NavSatFix topic not exist
error when running `rosrun mavros mavcmd takeoff 0 0 0 0 50`:
Request failed. Check mavros logs. ACK: 4

PS. Absolute height around 45m.
[image]

RPI3 and UAV with  GPS signal available.
[image]

When I run `rosrun mavros mavcmd set home 0 0 0` without GPS signal available, I will get the error:
Request failed. Check mavros logs. ACK: 4  

Hope this can help.

jamaruiz, 2018-10-02:
I'm having similar problems. Moving outdoors and getting a GPS signal does not change anything. Were you able to figure this out?

vooon, 2018-10-03:
@xup682u03 you have APM firmware, not PX4.

Try to set GUIDED mode first.

TSC21, 2019-05-31:
Solved in https://github.com/mavlink/mavros/pull/1128.

closed 2019-05-31 by TSC21