mavlink/mavros #1045: "No satellites" from "/diagnostics" mavros

Phadadev, 2018-06-02:
I placed my drone in the open area and led of pixhawk is green.
After I run this code
`roslaunch mavros px4.launch`
and then I run
`rostopic echo -n1 /diagnostics`
I found that
```
name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
```
But after I connect qgroundcontrol and exit program so I run mavros and check the "/diagnostics" again, I found the GPS show.
```
level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "16"
      - 
        key: "Fix type"
        value: "3"
```
I want to get the value from GPS without having to connect QGC. How can I do this?

### MAVROS version and platform

Mavros: -
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4

Version: lastest

### Diagnostics

```
header: 
  seq: 327
  stamp: 
    secs: 1527908144
    nsecs:  63593259
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "5340"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "240"
      - 
        key: "Tx sequence number:"
        value: "141"
      - 
        key: "Rx total bytes:"
        value: "132922"
      - 
        key: "Tx total bytes:"
        value: "133365"
      - 
        key: "Rx speed:"
        value: "305.000000"
      - 
        key: "Tx speed:"
        value: "320.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "415"
      - 
        key: "Frequency (Hz)"
        value: "0.960080"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "ArduPilotMega / ArduCopter"
      - 
        key: "Mode"
        value: "STABILIZE"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last dt (ms)"
        value: "3.333722"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "916.671332000"
      - 
        key: "Time offset (s)"
        value: "1527907227.282750130"

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 165 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 111

```

TSC21, 2018-06-02:
> But after I connect qgroundcontrol and exit program so I run mavros and check the "/diagnostics" again, I found the GPS show.

That's kinda akward. So you start and exit QGC and it then gets the GPS data? That seems a coincidence. Are you sure you didn't wait long enough to get the data? Also, QGC won't get the data from the autopilot with MAVROS taking control of the serial port, unless you have activated the the UDP forwarding to the GCS.

Phadadev, 2018-06-02:
I'm pretty sure about that because I tested more than ten times. And wait almost 20 minutes for each time.Is it possible that I was using M8P? But I have not  connected to base(rtk).If not connected to base will it work like a normal GPS?

vooon, 2018-06-02:
Not sure, but that looks logical to fallback to normal gps.
Two things are possible:

1. GCS does explicit setup of streams, so position fix messages start to stream.
   In that case try `rosrun mavros mavsys rate --all 10`, if that helps then you should configure stream rate parameters.
2. M8P waits some data for rtk and return 0-sats.

Phadadev, 2018-06-04:
@vooon  Oh,It's work.
Thank you so much.

Mulukulum, 2024-09-17:
Thanks this worked for me as well!