mavlink/mavros #1047: error with "rosrun mavros mavsys mode -c OFFBOARD"

xup682u03, 2018-06-08:
I was using mavros on rpi3 and testing mavros .

connect to px4

```
 source ~/catkin_ws/devel/setup.bash
 roscore
 rosrun mavros mavros_node _fcu_url:=/dev/ttyUSB0:921600
```

----
### MAVROS version and platform

Mavros: 0.18.5
ROS: Kinetic
Ubuntu: 16.04 MATE

### Autopilot type and version

[  ] ArduPilot
[O] PX4

### Issue details

I can't change mode to OFFBOARD
`rosrun mavros mavsys mode -c OFFBOARD`

[image]

mavros log
[image]

PS.
I can run rosrun mavros mavsafety arm with motor rotating and rosrun mavros mavcmd set home 0 0 0 without error.
But i encountered errors when running takeoff and takeoffcur in mavcmd.

TSC21, 2018-06-08:
You are not connected to PX4. The modes above definetly tell that your autopilot system is APM, not PX4. The similar mode you want in APM is called GUIDED. If it is PX4 that you want to use. Then you need to upload the firmware to it first.

TSC21, 2018-06-08:
Also a reminder to check the docs on how to use OFFBOARD mode in PX4.

closed 2018-06-09 by mhkabir