mavlink/mavros #1050: Using fake_gps

TrishantRoy, 2018-06-17:
----

### Issue details

Hello,
I am using Pixhawk 2 with APM firmware. I am trying to use the fake_gps feature of mavros in gps-denied environment. I am getting the localised data of the quad. 

What I have done:
I am publishing that data in the /mavros/fake_gps/mocap/pose topic
I have set the GPS_TYPE parameter to 14.
What else needs to be done?

I tried the example code: https://dev.px4.io/en/ros/mavros_offboard.html
I changed to GUIDED mode since OFFBOARD isn't available.
On executing, the quad armed and disarmed only.

Data is being published to /mavros/global_position/global and /mavros/global_position/local topics.

Am i missing something to use setpoint?

Thank you.
 
### MAVROS version and platform

Mavros: 0.25.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot 3.5.5

### Node logs

```
roslaunch mavros apm.launch 

... logging to /home/iarc/.ros/log/1febaee0-721d-11e8-9249-d46d6dbd4f34/roslaunch-iarc-nuc-7785.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.113:43529/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 0
[... 568 lines cut ...]
    values: 
      - 
        key: "Voltage"
        value: "0.00"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"

```

### Check ID

```
rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 3999 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 129, 2, 133, 1, 136, 137, 24, 150, 152, 22, 27, 29, 30, 193, 33, 35, 36, 165, 42, 173, 178, 182, 62, 65, 74, 77, 163, 100, 241, 116, 125

```

TSC21, 2018-06-17:
Have you checked the plugin configuration: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L161-L182 ?

TSC21, 2018-06-17:
> I tried the example code: https://dev.px4.io/en/ros/mavros_offboard.html
I changed to GUIDED mode since OFFBOARD isn't available.
>On executing, the quad armed and disarmed only.

This has nothing to do with this issue itself. If you have questions on how to use GUIDED mode with APM, you should open a new issue or ask in the APM forums.

>Data is being published to /mavros/global_position/global and /mavros/global_position/local topics.

What data? Your data? Or the data coming from the FCU?

>Am i missing something to use setpoint?

Again, `fake_gps` plugin only matters for estimation, not control. So, two separate issues.

TrishantRoy, 2018-06-17:
Thank you for the reply @TSC21 

I had edited the apm_config.yaml file. No need to edit the px4_config.yaml, right? 
```#fake_gps
fake_gps:
  # select data source
  use_mocap: true         # ~mocap/pose
  mocap_transform: false  # ~mocap/tf instead of pose
  use_vision: false       # ~vision (pose)
  # origin (default: Zürich)
  geo_origin:
    lat: 0        # latitude [degrees]
    lon: 0        # longitude [degrees]
    alt: 0            # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 2.0
  epv: 2.0
  satellites_visible: 5   # virtual number of visible satellites
  fix_type: 3             # type of GPS fix (default: 3D)
  tf:
    listen: false
    send: false           # send TF?
    frame_id: "map"       # TF frame_id
    child_frame_id: "fix" # TF child_frame_id
    rate_limit: 10.0      # TF rate
  gps_rate: 5.0           # GPS data publishing rate
```
Doubt: What other parameters to change?
1. Should the lat and lon be changed to my current location as well?

> What data? Your data? Or the data coming from the FCU?
FCU probably. Realised later that it wasn't data from localisation.
Is there a way to check whether the localisation data is being used by the Pixhawk i.e. is the estimation working? I will check the control part later then.

> Again, fake_gps plugin only matters for estimation, not control. So, two separate issues.
Yeah. Got your point.

TSC21, 2018-06-18:
> I had edited the apm_config.yaml file. No need to edit the px4_config.yaml, right?

You are using APM right? So, just APM is needed.

> Should the lat and lon be changed to my current location as well?

You can but not mandatory. This is just fake coordinates.

> Is there a way to check whether the localisation data is being used by the Pixhawk i.e. is the estimation working?

You can use both `mavros/global_position/global` and `mavros/local_position/local`. The first for LLA and the second for local position.

TrishantRoy, 2018-07-14:
Not completely actually. Have kept this aside for sometime.

I would suggest to not close the issue currently. But if required, close it. I will post queries related to this in a new post later.

Thanks.

pranavn248, 2018-08-06:
Hey, I seem to have some similar troubles with changing my home GPS position as well. I'm using PX4, and I made the necessary changes to the px4_config.yaml file (i.e.. use_mocap = false and changed LLA) and similar changes to the fake_gps.cpp plugin. I also changed MAVLINK_HILGPS to 1. But the PX4 doesn't seem to be responding to these, am I missing something?

closed 2018-10-16 by TSC21

px4Li, 2018-12-29:
@trishantroy Do you have any progress?

trishantroy, 2018-12-29:
I used the External Position Estimate feature to solve the issue I was facing.

px4Li, 2018-12-30:
@trishantroy That's great. 
Could you please tell me how to solve this:
  [ INFO] [1529234131.249046792]: HP: requesting home position
I am able to use vision position estimate to fly drone with position mode  but when I switch to mission mode that always be rejected. I think that is requesting home position. I saw my mavros message it has plugged in fake_gps, but why it always request home position?

Should I open a new issue for it?

TSC21, 2018-12-30:
> @trishantroy That's great.
> Could you please tell me how to solve this:
> [ INFO] [1529234131.249046792]: HP: requesting home position
> I am able to use vision position estimate to fly drone with [...]

@px4Li please do not continue the discussion on a closed issue. You have two solution for your problem, considering you are using PX4:
1. Use `fake_gps` plugin and send local position estimation as GPS data;
2. Use the normal vision pipeline but, on the PX4 side, use the LPE estimator instead of EKF2, and set the `LPE_FAKE_ORIGIN` parameter to `1`.