mavlink/mavros #1052: Cannot Send Data To FCU

MaMadDl, 2018-06-20:
### Issue details

Hi!
After Migrating To ROS Melodic On Ubuntu 18.04(ARM) I'm Unable To Send Data(Optical_Flow_RAD) Or Services(Arming ,SetMode) To PX4 Flightcontroller But I Can Get Data From It.

Installed Mavros With Apt-get.

### MAVROS version and platform

Mavros: 0.25.1
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.6.5 V1_default

### Node logs
```
started roslaunch server http://odroid:40703/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: True
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 334 lines cut ...]
      - 
        key: "Last dt (ms)"
        value: "0.000000"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "0.000000000"
      - 
        key: "Time offset (s)"
        value: "0.000000000"
```
### Check ID

```
odroid@odroid:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 6376 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 0, 83, 36, 230, 1, 105, 74, 2, 140, 141, 111, 76, 241, 147, 245, 30, 31

```

vooon, 2018-06-20:
Check px4 companion parameter. Try to use mavlink v1.0.

MaMadDl, 2018-06-21:
I'm Using Onboard Mavlink channel Which I started In extras.txt 
`mavlink start -d /dev/ttyS2 -b 921600 -m onboard` 
tried With config mode also with no success 
also it is Version 1 as i can see in QGC `MAV_PROTO_VER` is set to `deafult To 1 Switch to 2 If GCS sends Version 2` and also In NSH I have :
```
instance #1:
        mavlink chan: #1
        no telem status.
        flow control:   OFF
        rates:
        tx: 16.911 kB/s
        txerr: 0.000 kB/s
        rx: 0.000 kB/s
        rate mult: 1.000
        accepting commands: YES
        MAVLink version: 1
        transport protocol: serial (/dev/ttyS2 @921600)
```

vooon, 2018-06-21:
I mean on mavros side (param `~fcu_protocol=v1.0`).

bramsvs, 2018-06-21:
It looks like I'm having the same issue. I can receive `/mavros/mocap/pose` messages, but not `/mavros/vision_pose/pose` messages. Also Ubuntu 18.04/ROS Melodic. setting `fcu_protocol:=v1.0`.

MaMadDl, 2018-06-23:
Sorry For Late response.

My Mavlink protocol is Version 1 In Mavros Side Too.
`<arg name="fcu_protocol" default="v1.0" />`

also here's Debug log from mavconn
```
[DEBUG] [1529752810.115873808]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 133
[DEBUG] [1529752810.121896379]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 134
[DEBUG] [1529752810.125032436]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 135
[DEBUG] [1529752810.128008950]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 136
[DEBUG] [1529752810.130895256]: serial0: recv: v1.0   OK Message-Id: 105 [62 bytes] IDs: 1.1 Seq: 137
[DEBUG] [1529752810.131852136]: serial0: recv: v1.0   OK Message-Id: 31 [32 bytes] IDs: 1.1 Seq: 138
[DEBUG] [1529752810.132036428]: serial0: recv: v1.0   OK Message-Id: 32 [28 bytes] IDs: 1.1 Seq: 139
[DEBUG] [1529752810.133867229]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 140
[DEBUG] [1529752810.136959161]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 141
[DEBUG] [1529752810.140084301]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 142
[DEBUG] [1529752810.149308013]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 143
[DEBUG] [1529752810.152950655]: serial0: recv: v1.0   OK Message-Id: 105 [62 bytes] IDs: 1.1 Seq: 144
[DEBUG] [1529752810.153971369]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 145
[DEBUG] [1529752810.154198536]: serial0: recv: v1.0   OK Message-Id: 31 [32 bytes] IDs: 1.1 Seq: 146
[DEBUG] [1529752810.154461329]: serial0: send: OPTICAL_FLOW_RAD:
  time_usec: 1529752810109319
  sensor_id: 0
  integration_time_us: 1529752754
  integrated_x: 1.24653
  integrated_y: 1.27656
  integrated_xgyro: -0.00616302
  integrated_ygyro: -0.00102938
  integrated_zgyro: -0.204706
  temperature: 100
  quality: 250
  time_delta_distance_us: 1529752754
  distance: 2

[DEBUG] [1529752810.156146796]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 147
[DEBUG] [1529752810.164846547]: serial0: recv: v1.0   OK Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 148
[DEBUG] [1529752810.164998714]: serial0: recv: v1.0   OK Message-Id: 32 [28 bytes] IDs: 1.1 Seq: 149

```

and upon calling every service with result response i get the result 4
```
odroid@odroid:~$ rosservice call /mavros/cmd/arming "value: true" 
success: False
result: 4
```

MaMadDl, 2018-06-28:
Ok Solved The Problem.
It Was My USB2UART Which Couldn't Send Data On 921600 baudrate But Can Read On it (Weird!)

closed 2018-06-28 by MaMadDl

vooon, 2018-07-02:
@MaMadDl can you please provide info about this chip (VID/PID, manufacturer, name)? That may be helpful for other users.

weixuanzhang, 2019-04-29:
@MaMadDl can you please say how you solved the problem? We encountered a similar issue and would like to know how you solved it. Much appreciated!

MaMadDl, 2019-05-01:
On Newer Version Of Mavros_extras There's a Topic to send Optical_Flow_Rad message topic to FCU. you can either enable it in pluginlist or launch px4flow.launch in Mavros_extars

weixuanzhang, 2019-05-08:
Same issue here. We also found out the USB2UART is physically disconnected