mavlink/mavros #1055: Problem with cmd_vel for steering the rover

chouaibharik, 2018-07-12:
I was using /mavros/rc/override on a pixhawk 1 in order to control the rover via an external pc, and this worked just fine. However, i needed to upgrade to the pixhawk 2 (the first one died), and now the RCoverride is no longer supported so i try now to control the rover by publishing to /mavros/setpoint_velocity/cmd_vel on GUIDED mode (otherwise the rover wouldn't respond).

Nonetheless, weird things are happening:

1- Sending +0.1 on twist.linear.x makes the robot moves forward (which is normal), but sending the same value on twist.linear.y  makes the robot moves forward as well (which is not logical since this is not a holonomic platform, so no direction on the Y axis should be possible).

2- Sending -0.1 on twist.linear.x does not make the robot to move backward, but forward as well???

3- The angular velocity (twist.angular.z) seems to work fine with speed and direction (the robot is a bit shaking when turning on its center of gravity).

Any ideas about this?

Thank you in advance

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: latest_from_apt-get
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 3.4.0

### Node logs

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
[... 424 lines cut ...]
      - 
        key: "Frequency (Hz)"
        value: "9.613204"
      - 
        key: "Last RTT (ms)"
        value: "16.270802"
      - 
        key: "Mean RTT (ms)"
        value: "16.182200"
      - 
        key: "Last remote time (s)"
        value: "21004.780771000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

### Check ID

OK. I got messages from 1:1.

---
Received 1112 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 129, 2, 22, 1, 137, 128, 152, 147, 150, 24, 27, 29, 30, 32, 33, 34, 35, 36, 165, 193, 42, 178, 182, 65, 74, 163, 111, 241, 116, 124, 125

vooon, 2018-07-12:
Unfortunately i do not see anything in thins logs, except strange FW version reports.
That may be caused by some commands or reboots. Unlikely that this is because reboot, because in that case drone would be uncontrollable for some seconds.
But try to check wiring.

Also how autopilot is connected to mavros? High Round-Trip Time looks weird.

chouaibharik, 2018-07-13:
Thanks for your reply. I am using a rover (which can be as well called a drone, but just for avoiding confusion :D). The wiring is fine, since the manual + auto (WP from MP) work just fine, in addition i am able to communicate with the pixhawk, so i doubt that the wiring is the problem.

I have tried to connect the autopilot through the usb port on the pixhawk, and the telemetry port, but the results are the same.

Is there a way to send direct commands to the motors? something similar to rcoverride that was before?

vooon, 2018-07-13:
You may try COMMAND_LONG with [DO_SET_SERVO][1].

[1]: https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO

chouaibharik, 2018-07-13:
I am trying to send this command (to turn the right wheel attached to output 1 for 10 cycles of 10s):

rosservice call /mavros/cmd/command "{broadcast: true, command: 184, confirmation: 0, param1: 1.0, param2: 1600.0, param3: 10.0, param4: 10.0, param5: 0.0, param6: 0.0, param7: 0.0}" 

Nothing is happening, should the rover be in GUIDED mode when using DO_SET_SERVO? or just manual mode?

khancyr, 2018-07-13:
DO_SET_SERVO isn't implemented on on ArduPilot Rover (you write you are using PX4 but speak about Guided).
RC override should work in non guided mode but you need to set SYSID_MYGCS to 1 .
About setpoint velocity, the default frame is Local_ned, so you are sending a speed VECTOR in NED frame so x isn't front and y isn't lateral, that is doesn't match the default (and wrong) behavior from ROS that said the x is always front. And as we calculate the velocity vector, we lost the reverse movement. Changing the frame to body_frame should change this behavior .

chouaibharik, 2018-07-13:
I have already tried to set SYSID_MYGCS to 1 but the RC override was still present. I will try to change the frame and see if this will solve the problem.

hmchung, 2018-07-16:
Hi guys. I got into the same issue:

1) I ran mavros with companion link 921600
2) I got the same warning message "TM : RTT too high for timesync"
3) Interestingly, with this, I could not send setpoints and could not enter offboard mode
4) Most amazingly, the issue goes away when I plug in the micro USB. But this is just a very bad hack. 

Waiting for more comments.

chouaibharik, 2018-07-17:
Thank you for your feedback. I have other projects going on, but for now i just used an arduino with a 5v switch in order to choose between who takes control of the ESCs (pc or Pixhawk)...It is a quick and dirty solution for my autonomous visual tracking and other schemes, but it works just fine.

I will try your trick soon (hopefully)

isakdiaz, 2018-09-11:
You have to set the MAV_FRAME to the local bodyframe so that the commands are processed in that coordinate system.

Just run:
rosservice call /mavros/setpoint_velocity/mav_frame "mav_frame: 8" 

Here are more frames to try:
https://github.com/mavlink/mavros/blob/master/mavros_msgs/srv/SetMavFrame.srv

chouaibharik, 2019-05-03:
Hello guys,

I finally figured it out, and i can now control the velocity of the rover directly through mavros.

First, the rover should be in GUIDED mode. To do this, whether you set this though your RC via MP or other GUI, or you simply use mavros. In another terminal (besides the one running mavros with the adequte autopilot, for me i used apm2.launch):

`rosrun mavros mavsys -c GUIDED`

Once the mode is changed, you can try publishing on the following topic: /mavros/setpoint_velocity/cmd_vel (for the rover, linear velocity on the x, and angular around z). You should be able to see the rover moving for few seconds.

`rostopic pub /mavros/setpoint_velocity/cmd_vel geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
twist:
  linear:
    x: 0.5
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.3" 
`

khancyr, 2019-05-03:
@chouaibharik Thanks for the feedback ! I glad you manage to make it work .
@TSC21 this issue could be close

closed 2019-05-03 by TSC21

githinkgp, 2019-05-21:
@chouaibharik When you got it working what did you set your setpoint/velcoity/mav_frame to? Is it LOCAL_NED or BODY_NED?

I am also curious that in the TwistStamped message that you are publishing you have left the frame id slot. Did you leave it blank in your implementation as well?

`rostopic pub /mavros/setpoint_velocity/cmd_vel geometry_msgs/TwistStamped "header: seq: 0 stamp: secs: 0 nsecs: 0 **frame_id: ''** twist: linear: x: 0.5 y: 0.0 z: 0.0 angular: x: 0.0 y: 0.0 z: 0.3"`

Please let me know. 
Thank you

chouaibharik, 2019-05-21:
I haven't set the mav frame, and I left the frame id blank on the implementation as well.

githinkgp, 2019-05-21:
@chouaibharik Thank you for the quick reply.
I have been using PX4 and not ardupilot. The OFFBOARD mode in PX4 works the same way as GUIDED in AP.
This is what I have been trying.
1. roslaunch mavros px4.launch, with the timesync_rate disabled to avoid the WARN: RTT too high for timesync.
2.     The tf now looks like as attached.
3.     I then run a node to publish Twist message on mavros/setpoint_velocity/cmd_vel_unstamped at 10 Hz. I defined each of the linear and angular velcoities as ros paramters and initialized them to 0.
4.     Once I have the above node running, I switch the mode to OFFBOARD and then arm the vehicle. Both these happen successfully.
5.     Now I use the rosparam command to change the values of linear and angular velocity being published.

I do this and still get no response from the rover.
I have tried publishing to the sepoint_velocity/cmd_vel topic in the same way you did but still couldn't get it to work. I tried giving x, y and z velcities but nothing worked.

I don’t know where I am going wrong.

Any suggestions would be helpful.
Thank you.

These are my mavros params:
CLEAR PARAMETERS

    /mavros/

PARAMETERS

    /mavros/cmd/use_comp_id_system_control: False
    /mavros/conn/heartbeat_rate: 1.0
    /mavros/conn/system_time_rate: 1.0
    /mavros/conn/timeout: 10.0
    /mavros/conn/timesync_rate: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
    /mavros/distance_sensor/hrlv_ez4_pub/id: 0
    /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
    /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
    /mavros/distance_sensor/laser_1_sub/id: 3
    /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
    /mavros/distance_sensor/laser_1_sub/subscriber: True
    /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
    /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
    /mavros/distance_sensor/lidarlite_pub/id: 1
    /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
    /mavros/distance_sensor/lidarlite_pub/send_tf: True
    /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
    /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
    /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
    /mavros/distance_sensor/sonar_1_sub/id: 2
    /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
    /mavros/distance_sensor/sonar_1_sub/subscriber: True
    /mavros/fake_gps/eph: 2.0
    /mavros/fake_gps/epv: 2.0
    /mavros/fake_gps/fix_type: 3
    /mavros/fake_gps/geo_origin/alt: 408.0
    /mavros/fake_gps/geo_origin/lat: 47.3667
    /mavros/fake_gps/geo_origin/lon: 8.55
    /mavros/fake_gps/gps_rate: 5.0
    /mavros/fake_gps/mocap_transform: True
    /mavros/fake_gps/satellites_visible: 5
    /mavros/fake_gps/tf/child_frame_id: fix
    /mavros/fake_gps/tf/frame_id: map
    /mavros/fake_gps/tf/listen: False
    /mavros/fake_gps/tf/rate_limit: 10.0
    /mavros/fake_gps/tf/send: False
    /mavros/fake_gps/use_mocap: True
    /mavros/fake_gps/use_vision: False
    /mavros/fcu_protocol: v2.0
    /mavros/fcu_url: /dev/ttyUSB0:57600
    /mavros/gcs_url:
[... 171 lines cut ...]
[ INFO] [1558464790.271138714]: Plugin vision_pose_estimate loaded
[ INFO] [1558464790.363144037]: Plugin vision_pose_estimate initialized
[ INFO] [1558464790.364172772]: Plugin vision_speed_estimate loaded
[ INFO] [1558464790.409601010]: Plugin vision_speed_estimate initialized
[ INFO] [1558464790.411010145]: Plugin waypoint loaded
[ INFO] [1558464790.460253103]: Plugin waypoint initialized
[ INFO] [1558464790.460520045]: Plugin wheel_odometry blacklisted
[ INFO] [1558464790.461924700]: Plugin wind_estimation loaded
[ INFO] [1558464790.468775007]: Plugin wind_estimation initialized
[ INFO] [1558464790.469057979]: Autostarting mavlink via USB on PX4
[ INFO] [1558464790.469322330]: Built-in SIMD instructions: ARM NEON
[ INFO] [1558464790.469609206]: Built-in MAVLink package version: 2019.2.2
[ INFO] [1558464790.469870709]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1558464790.470158769]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1558464790.677648605]: RC_CHANNELS message detected!
[ INFO] [1558464791.182722040]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1558464791.267209127]: RC_CHANNELS message detected!
[ INFO] [1558464791.502010553]: IMU: High resolution IMU detected!
[ INFO] [1558464792.227014851]: VER: 1.1: Capabilities 0x00000000000024ef
[ INFO] [1558464792.227680560]: VER: 1.1: Flight software: 01060500 (3098c77c9168553c)
[ INFO] [1558464792.228063104]: VER: 1.1: Middleware software: 01060500 (3098c77c9168553c)
[ INFO] [1558464792.228373433]: VER: 1.1: OS software: 000000c0 (0000000000000000)
[ INFO] [1558464792.228663732]: VER: 1.1: Board hardware: 00000011
[ INFO] [1558464792.228959311]: VER: 1.1: VID/PID: 26ac:0011
[ INFO] [1558464792.229246955]: VER: 1.1: UID: 3236511737353639

chouaibharik, 2019-05-21:
Do you see the topic /mavros/rc/override on the topics list?

githinkgp, 2019-05-21:
Yes and nothing is published on it.

chouaibharik, 2019-05-21:
Of course not, it is a topic where you emulate a RC, so naturally nothing is published on that topic. Try to publish on it (just use rostopic pub command), while choosing the right channel for your motors.

kissw, 2020-02-03:
@chouaibharik  @githinkgp  
I have the same problem.
I want to reverse my vehicle, so I tried to the setpoint_velocity/cmd_vel/twist/linear/x value to negative, But the vehicle will not reverse. 
Strangely, changing the linear/y value causes the vehicle to move forward and steering. 

Also, changing the angular/z value will not be steering, when the linear/x value is set to -1, the vehicle rotates clockwise in place as it moves forward.

The mode is the "Offboard" mode and the mavros/state/guided value is true.
I tried to change the mav_frame value, but it wasn't resolved.

If you've solved it, could you explain it in more detail?

chouaibharik, 2020-02-03:
> @chouaibharik @githinkgp
> I have the same problem.
> I want to reverse my vehicle, so I tried to the setpoint_velocity/cmd_vel/twist/linear/x value to negative, But the vehicle will not reverse. [...]

Have you tried the steps i explained in https://github.com/mavlink/mavros/issues/1055#issuecomment-489127132 ?

kissw, 2020-02-03:
[image]
When I typing this command, result is this.

[image]
I connected the mavros using` roslaunch px4 apm2.launch` and entered Guided mode.
But the results are the same.

chouaibharik, 2020-02-03:
@kissw  First, are you sure your autopilot is a px4 not an APM?. Second, you will need to add the guided mode on MP (for my case, i have a three way switch that i can chose between Stabilize, Guided and, Auto). Then the switch will be on the Guided position in order to be able to pass the commands to the autopilote.

Jaeyoung-Lim, 2020-02-03:
@kissw I enabled velocity control for rover on px4. However, this was only enabled after PX4 v1.10 what version are you using?

kissw, 2020-02-03:
@chouaibharik 
First, Yes, I'm using PX4. 
Second, Actually I'm testing in simulation environment... What is the MP?

@Jaeyoung-Lim 
I checked, it's a master branch. I'll change the firmware version to version 1.10, retest and let you know.

Jaeyoung-Lim, 2020-02-03:
@kissw Which master are you on? If you are on master, you should be better of than 1.10 since the actuator controls should also be included

chouaibharik, 2020-02-03:
@kissw  MP is Mission Planner.

kissw, 2020-02-03:
@Jaeyoung-Lim I tried to v1.11.0-beta1, but I still this ploblem.
Here's the link that recorded my test.
[https://youtu.be/SJXJnuGTxIo](url)

Jaeyoung-Lim, 2020-02-03:
@kissw Oh, sorry. The vehicle will not reverse since it is constrained.

you can change that here: https://github.com/PX4/Firmware/blob/master/src/modules/rover_pos_control/RoverPositionControl.cpp#L301

kissw, 2020-02-03:
@Jaeyoung-Lim 
Thank you so much!
This Rover seems to be able to do something like path planning, but I want a simple Rover that can be controlled using angular.z and linear.x velocities.
Is there the model or branch that i want?

Jaeyoung-Lim, 2020-02-03:
@kissw Currently for offboard mode, the mavlink messages that are supported are listed in the [docs](https://docs.px4.io/master/en/flight_modes/offboard.html), which mavros sends.

It does not support angular velocities, although I was planning to add it. You can still use linear velocity in both body and local frame or use actuator controls directly.

kissw, 2020-02-03:
@Jaeyoung-Lim 
Modification of the code have caused the Rover able to make reverse. But not only did the Rover reverse, it even steered to a 180 degree turn. i think it seems to need to be complemented. Anyway, Thank you very much for your reply.

Jaeyoung-Lim, 2020-02-03:
@kissw That is why it was constrained. You need additional logic to handle reverse