mavlink/mavros #1058: Keep dropping Message and said：TX queue overflow， when publish topic to mavros/setpoint_raw/local

eleboss, 2018-07-18:
Dear team,
Could you guys give some help or advice about this. 
I have no idea where to debug this, I am using same code, but this error just appear  in second day, seems strange.
----

### Issue details
I am using OFFBOARD mode to control the Drone to fly indoor, I have depth sensor and visual tag as feedback and using speed closeloop control. But when I publish msg to  /mavros/setpoint_raw/local at 100hz for several second, the error appeared,  and seems the drone can not receive any msgs, I test several publish rate and different pixhawk(2 and 4) the error still exist.
I am complete lost. please give a direction. 

[ERROR] [1531909356.890979783]: 0: DROPPED Message SET_POSITION_TARGET_LOCAL_NED: MAVConnSerial::send_message: TX queue overflow

My launch procedure:
terminal1:  roslaunch mavros px4.launch 
terminal2:  rosrun offboard drone_control.py

My px4.launch
```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- example launch script for PX4 based FCU's -->

	<arg name="fcu_url" default="/dev/ttyACM0:921600" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
		<arg name="fcu_protocol" value="$(arg fcu_protocol)" />
		<arg name="respawn_mavros" default="$(arg respawn_mavros)" />
	</include>
</launch>
```

the main loop of my control code
```
rospy.init_node('drone_control')
PIDinit()

rc_in = rospy.Subscriber('mavros/rc/in', RCIn, callback_rc, queue_size=1)
sub_state = rospy.Subscriber('mavros/state', State, callback_state, queue_size=1)
subodom = rospy.Subscriber('mavros/local_position/odom', Odometry, callback_odom)
subdepth = rospy.Subscriber('depth', Odometry, callback_depth)
position_pub = rospy.Publisher('mavros/setpoint_raw/local', PositionTarget, queue_size=1)
#rospy.spin()

rate = rospy.Rate(100)  
while not rospy.is_shutdown():

    vel_control = TwistStamped()
    pos_control = PoseStamped()

    scale_x = 0.4
    scale_y = 0.4
    scale_z = 1
    scale_yaw = 1

    pid_x.update(feedback_x)
    output_x = pid_x.output * scale_x
    feedback_x = pos_fuse_x
[... 499 lines cut ...]
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:921600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "44"
      - 
        key: "Frequency (Hz)"
        value: "3.507350"
      - 
        key: "Last RTT (ms)"
        value: "6450.054686"
      - 
        key: "Mean RTT (ms)"
        value: "281.613853"
      - 
        key: "Last remote time (s)"
        value: "702.575705000"
      - 
        key: "Estimated time offset (s)"
        value: "1531909221.221841335"
---

```

vooon, 2018-07-19:
Look at diag entry "mavros: FCU connection", when you got overflow.
Check rate of your setpoints, on sub 1M baud 100 Hz should be ok.
But you somehow generate much more messages than port may handle.
TX queue have 1k limit, which in normal case newer hit (usually 10 entries max).

Other possible thing, that your UART adaptor cannot handle 921600 and effective bandwidth limited by something else.

eleboss, 2018-07-19:
YES!!!!!!!!!!!!
You are right about this！
I checked the publish rate of the topic, and found it become 1000HZ, not 100 HZ, and then I found there is a indent error in my code that failed the rate.sleep(), I modified it ,and it work as usual !

@vooon Thank you very much!

closed 2018-07-19 by eleboss

Satano1991, 2020-06-29:
Hi guys, 
I guess that I have same problem like you, eleboss... I changed FCU on 921600 but I would like to know where did you find publish rate! 
Please, can you help me? THX