mavlink/mavros #1061: can't build trajectory.cpp

yc718764666, 2018-07-19:
When compile the mavros I encountered the problem, I can not compile successfully. When build trajectory.cpp, if I comment the trajectory.cpp in Cmakelist, I compile it successfully.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 0.26
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

khancyr, 2018-07-19:
Hello,
Can you give us the compilation error in order to work on a correct fix ?

yc718764666, 2018-07-19:
OK, But it may be a little long
[ 93%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/rc_io.cpp.o
[ 94%] Building CXX object mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/vision_speed_estimate.cpp.o
[ 94%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/command.cpp.o
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:136:53: error: ‘mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS’ has not been declared
  auto fill_points_unused_path(mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS & t, const size_t i) {
                                                     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:146:52: error: ‘mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS’ has not been declared
  void fill_points_all_unused(mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS &t, const size_t i) {
                                                    ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:164:85: error: ‘TRAJECTORY_REPRESENTATION_WAYPOINTS’ in namespace ‘mavlink::common::msg’ does not name a type
  void fill_msg_position(geometry_msgs::Point &position, const mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS &t, const size_t i)
                                                                                     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:173:87: error: ‘TRAJECTORY_REPRESENTATION_WAYPOINTS’ in namespace ‘mavlink::common::msg’ does not name a type
  void fill_msg_velocity(geometry_msgs::Vector3 &velocity, const mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS &t, const size_t i)
                                                                                       ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:182:95: error: ‘TRAJECTORY_REPRESENTATION_WAYPOINTS’ in namespace ‘mavlink::common::msg’ does not name a type
  void fill_msg_acceleration(geometry_msgs::Vector3 &acceleration, const mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS &t, const size_t i)
                                                                                               ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:284:86: error: ‘mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS’ has not been declared
  void handle_trajectory(const mavlink::mavlink_message_t *msg, mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS &trajectory)
                                                                                      ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp: In member function ‘auto mavros::extra_plugins::TrajectoryPlugin::fill_points_unused_path(int&, std::size_t)’:
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:137:5: error: request for member ‘vel_x’ in ‘t’, which is of non-class type ‘int’
   t.vel_x[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:138:5: error: request for member ‘vel_y’ in ‘t’, which is of non-class type ‘int’
   t.vel_y[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:139:5: error: request for member ‘vel_z’ in ‘t’, which is of non-class type ‘int’
   t.vel_z[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:140:5: error: request for member ‘acc_x’ in ‘t’, which is of non-class type ‘int’
   t.acc_x[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:141:5: error: request for member ‘acc_y’ in ‘t’, which is of non-class type ‘int’
   t.acc_y[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:142:5: error: request for member ‘acc_z’ in ‘t’, which is of non-class type ‘int’
   t.acc_z[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:143:5: error: request for member ‘vel_yaw’ in ‘t’, which is of non-class type ‘int’
   t.vel_yaw[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp: In member function ‘void mavros::extra_plugins::TrajectoryPlugin::fill_points_all_unused(int&, std::size_t)’:
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:147:5: error: request for member ‘pos_x’ in ‘t’, which is of non-class type ‘int’
   t.pos_x[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:148:5: error: request for member ‘pos_y’ in ‘t’, which is of non-class type ‘int’
   t.pos_y[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:149:5: error: request for member ‘pos_z’ in ‘t’, which is of non-class type ‘int’
   t.pos_z[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:151:5: error: request for member ‘vel_x’ in ‘t’, which is of non-class type ‘int’
   t.vel_x[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:152:5: error: request for member ‘vel_y’ in ‘t’, which is of non-class type ‘int’
   t.vel_y[i] = NAN;
     ^
/home/yc/catkin_ws3/src/mavros/mavros_extras/src/plugins/trajectory.cpp:153:5: error: request for member ‘vel_z’ in ‘t’, which is of non-class type ‘int’
   t.vel_z[i] = NAN;
     ^
[... cut ...]
mavros/mavros_extras/CMakeFiles/mavros_extras.dir/build.make:302: recipe for target 'mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/trajectory.cpp.o' failed
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/trajectory.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
[ 95%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/local_position.cpp.o
[ 95%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/global_position.cpp.o
[ 95%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_position.cpp.o

TSC21, 2018-07-19:
There's a mavlink ROS release done that fixes this. Check https://github.com/mavlink/mavlink-gbp-release. If you don't want to wait till it gets to the Linux deb pkg list, you will have to clone it to your ws and recompile.

closed 2018-07-19 by TSC21