mavlink/mavros #1063: Altitude of global position doesn't match mavlink message #33

DesmondZhong, 2018-07-23:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, this is the message published to the `mavros/global_position/global` topic.

[image]

Notice latitude, longitude and altitude value. Now, here is the mavlink message obtained by MAVLink Inspector.

[image]

The latitude and the longitude are the same as the `mavros/global_position/global` topic. However the altitude is different. There is a difference of about 48m. 

I wonder why `mavros/global_position/global` topic isn't getting the correct altitude. The previous screenshots are from SITL with Gazebo. The home altitude posted on `mavros/home_position/home` is different from that of the home position message directly from MAVLink inspector too. From the output of mavros_node, the home altitude posted on `mavros/home_position/home` is incorrect and the home altitude directly from MAVLink inspector is correct. 

I also tested in a real world environment. There is still a difference between the two altitude values. Is this a bug or did I miss anything like altitude offset?

Thank you very much!

### MAVROS version and platform

Mavros: 0.26.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

TSC21, 2018-07-23:
Because MAVROS converts from the AMSL to the altittude above the WGS-84 datum ellipsoid, which is the ROS std for NavSat msgs. That's one of the main purposes of GeographicLib and it's documented on MAVROS readme file.

closed 2018-07-23 by TSC21

TSC21, 2018-07-24:
Yes. Check the `altitude` plugin.

DesmondZhong, 2018-07-24:
Thanks! It's not in the ROS documentation and I missed it.